cvar-upm/aerostack #1: Inaccessible submodule

andre-nguyen, 2016-02-22:
There's something blocking my from updating my submodules. I think it's this one: https://bitbucket.org/joselusl/quadrotor_stack_configs is it maybe private?

joselusl, 2016-02-22:
Fixed! Sorry!
They might be others submodule with the same problem. Please, let me know and I will make them public!

andre-nguyen, 2016-02-22:
This one might be out of your control though:

https://bitbucket.org/ramon_suarez_fernandez/robotsimulatorrvizrosmodule.git

joselusl, 2016-02-22:
I have fixed one that was under my control, the other one I am working on fixing it.
In the meantime, you can remove this one: https://bitbucket.org/ramon_suarez_fernandez/robotsimulatorrvizrosmodule.git from your git submodule's lists
It is not needed for the main functionality, it is just an IARC Mission 7 related one.

andre-nguyen, 2016-02-22:
Ill try to see if there isn't some kind of bulk way to list all the ones I don't have access to, instead of doing this as we go :sweat_smile:

joselusl, 2016-02-22:
Perfect!
I have already fixed the other one ;)

andre-nguyen, 2016-02-22:
# Summary
So I think these are the still private repos methodology below.

- [ ] https://bitbucket.org/ramon_suarez_fernandez/robotlocalizationrosmodule.git
- [ ] https://bitbucket.org/Vision4UAV/dronecomputervisioniarc14.git
- [ ] https://bitbucket.org/jespestana/droneSimulator.githttps://bitbucket.org/joselusl/dronesimulatorrosmodule.git
- [ ] https://bitbucket.org/jespestana/dronesimulatorirobotcvrosmodule.git
- [ ] https://bitbucket.org/ramon_suarez_fernandez/robotsimulatorrosmodule.git
- [ ] https://bitbucket.org/ramon_suarez_fernandez/dronesimulatorrvizrosmodule.git
- [ ] https://bitbucket.org/csampedro/dronesupervisorrosmodule
- [ ] https://bitbucket.org/csampedro/dronesupervisorrosmodule_v2
- [ ] https://bitbucket.org/csampedro/dronemissionplanneremulatorrosmodule
- [ ] https://bitbucket.org/csampedro/dronemanagerofactionsrosmodule
- [ ] https://bitbucket.org/ramon_suarez_fernandez/dronemissionscheduleprocessor
- [ ] https://bitbucket.org/ramon_suarez_fernandez/dronemissionscheduleprocessorrosmodule
- [ ] https://bitbucket.org/joselusl/dronespeechrosmodule.git
- [ ] https://bitbucket.org/joselusl/droneleapmotiondronecontroluserinterfacerosmodule
- [ ] https://bitbucket.org/joselusl/dronecommunicationmanagerrosmodule
- [ ] https://bitbucket.org/joselusl/lib_levmar-2.6
- [ ] https://bitbucket.org/joselusl/droneobstacledistancecalculatorrosmodule.git
- [ ] https://bitbucket.org/joselusl/dronesoundrosmodule.git

# Methodology
Alright so I cleaned up the git submodule file to get

```
https://bitbucket.org/jespestana/driver_camera_ueye.git
https://bitbucket.org/jespestana/driver_camera_ueye_rosmodule.git
https://github.com/cvg/px-ros-pkg.git
https://bitbucket.org/jespestana/driver_px4flow_interface_rosmodule.git
https://bitbucket.org/joselusl/driverparrotardronerosmodule.git
https://bitbucket.org/joselusl/mav_tools.git
https://bitbucket.org/joselusl/asctec_drivers.git
https://bitbucket.org/ramon_suarez_fernandez/driverpelicanrosmodule.git
https://bitbucket.org/Vision4UAV/okto_driver.git
https://bitbucket.org/joselusl/driveroktorosmodule.git
https://bitbucket.org/joselusl/dronemsgsros.git
https://bitbucket.org/joselusl/dronemoduleros.git
https://bitbucket.org/jespestana/dronemoduleinterfaceros.git
https://bitbucket.org/joselusl/lib_cvgekf.git
https://bitbucket.org/jespestana/lib_cvgthread.git
https://bitbucket.org/jespestana/lib_cvgutils.git
https://bitbucket.org/joselusl/lib_newmat11.git
https://bitbucket.org/joselusl/lib_pose.git
https://bitbucket.org/joselusl/lib_pugixml.git
https://bitbucket.org/joselusl/referenceframes.git
https://bitbucket.org/joselusl/videopublisherrosmodule.git
https://bitbucket.org/jespestana/droneekfstateestimator.git
https://bitbucket.org/jespestana/droneekfstateestimatorrosmodule.git
https://bitbucket.org/joselusl/dronevisualmarkerslocalizer.git
https://bitbucket.org/joselusl/dronevisualmarkerslocalizerrosmodule.git
https://bitbucket.org/joselusl/referenceframesros.git
https://bitbucket.org/jespestana/lib_cvglogger.git
https://bitbucket.org/jespestana/lib_cvgloggerros.git
https://bitbucket.org/jespestana/droneloggerrosmodule.git
https://bitbucket.org/jespestana/dronepelicanloggerrosmodule.git
https://bitbucket.org/joselusl/droneyawcommanderrosmodule.git
https://bitbucket.org/joselusl/dronetrajectoryplanner.git
https://bitbucket.org/joselusl/dronetrajectoryplannerrosmodule.git
https://bitbucket.org/joselusl/droneobstacleprocessorvisualmarks.git
https://bitbucket.org/joselusl/droneobstacleprocessorvisualmarksrosmodule.git
https://bitbucket.org/ramon_suarez_fernandez/robotlocalizationrosmodule.git
https://bitbucket.org/jespestana/dronepbvspositionmidlevelcontroller.git
https://bitbucket.org/jespestana/dronespeedpositionmidlevelcontroller.git
https://bitbucket.org/jespestana/dronetrajectorycontroller.git
https://bitbucket.org/jespestana/dronetrajectorycontrollerrosmodule.git
https://bitbucket.org/jespestana/dronetrajectorystatemachine.git
https://bitbucket.org/jespestana/droneiarc14missionschedulerrosmodule.git
https://bitbucket.org/joselusl/dronemissionplanner.git
https://bitbucket.org/joselusl/dronemissionplannerrosmodule.git
[... 188 lines cut ...]
Pass: https://github.com/ros-drivers/leap_motion
Username for 'https://bitbucket.org': 
Password for 'https://bitbucket.org': 
Fail: https://bitbucket.org/joselusl/droneleapmotiondronecontroluserinterfacerosmodule
Username for 'https://bitbucket.org': 
Password for 'https://bitbucket.org': 
Fail: https://bitbucket.org/joselusl/dronecommunicationmanagerrosmodule
Pass: https://github.com/anqixu/ueye_cam
Pass: https://bitbucket.org/joselusl/quadrotor_stack_configs
Username for 'https://bitbucket.org': 
Password for 'https://bitbucket.org': 
Fail: https://bitbucket.org/joselusl/lib_levmar-2.6
Username for 'https://bitbucket.org': 
Password for 'https://bitbucket.org': 
Fail: https://bitbucket.org/joselusl/droneobstacledistancecalculatorrosmodule.git
Username for 'https://bitbucket.org': 
Password for 'https://bitbucket.org': 
Fail: https://bitbucket.org/joselusl/dronesoundrosmodule.git
Pass: https://bitbucket.org/hridaybavle/driverpixhawkrosmodule.git
Pass: https://github.com/mavlink/mavros.git
Pass: https://bitbucket.org/hridaybavle/driverlidarliteinterfacerosmodule
Pass: https://github.com/AutonomyLab/bebop_autonomy.git
Pass: https://bitbucket.org/zorana/driverbeboprosmodule.git
Pass: https://bitbucket.org/joselusl/aerostack_installer
```

joselusl, 2016-03-01:
solved!
I have removed some deprecated packages that were causing the problems and I have made public the other ones!
Thanks!

andre-nguyen, 2016-03-01:
Clean recursive clone now works without any issues

:smile_cat:

closed 2016-03-01 by andre-nguyen