cvar-upm/aerostack #3: asctec_autopilot node not found

shlapion, 2016-05-30:
Hi, 

I am trying to run the framework on an AscTec Pelican. 
I followed the launch sequence from the pelican_real_flight_instructions file.
When I run the launchfile asctec_driver_node I get the following error:
`ERROR: cannot launch node of type [asctec_autopilot/autopilot_node]: asctec_autopilot`
I have roscore started and also sourced the proper workspace.

Is there a specific AscTec SDK that I must flash the HL with?
I have tried both V2 and V3.
The OS on the Pelican is installed form the newest recovery image found on the AscTec wiki page.

joselusl, 2016-06-02:
Hi,

We are using the catkinized version of asctec_drivers: http://wiki.ros.org/asctec_drivers
Please read all the associated documentation in this package regarding to flashing the AscTec Pelican. Note that our catkinized version (https://bitbucket.org/joselusl/asctec_drivers) is going to differ when compiling.

Bests

hridaybavle, 2016-06-02:
Hi, 

The problem seems to be because the node is not being detected, not the SDK version atleast now. Can you check inside Aerostack/stack/droneDrivers/driversPlatforms/driverAsctecPelican/asctec_drivers if you have all the files regarding the package.

joselusl, 2016-06-30:
I have fixed the environmental variables that were not properly set.
Please update everything (using the installer ;) )
Bests

closed 2016-06-30 by joselusl