cvar-upm/aerostack #4: Invoking "make cmake_check_build_system" failed

AloshkaD, 2016-06-17:
Hi,

I have encountered several errors while installing the Aerostack on ubuntu 14.04. Some of these errors are caused by dead link on bitbucket.org. I've included two errors and their screenshots for convenience.  I appreciate your help with this!  
Note: Aerostack on bitbucket.org is locked

Here is Error 1

CMake Error at quadrotor_stack/stack/droneDrivers/driversPlatforms/driverPixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

Here is Error 2

-- Configuring incomplete, errors occurred!
See also "/home/alex/workspace/ros/quadrotor_stack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/alex/workspace/ros/quadrotor_stack_catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
 

[image]
[image]

.

joselusl, 2016-06-24:
Hi,
Sorry for the error.
It should be fixed now. Please download and install again.
Bests

AloshkaD, 2016-06-25:
Jose,

I really appreciate you taking the time to respond.
I've noticed that you did some changes days ago and I went ahead and tried to re-install it, but It kept giving the same error. Consequently, I deferred to the repository that I'd forked from the Aerostack and finally managed to install everything by following the "/installation/installation_instructions.txt"except for the Human Machine Interface repositories from Yolanda. That one was giving errors due to missing c++ header libraries. I've contacted Yolanda for help but the deleted the repository all together. My question is, is there an alternative for the hmi_cvg_stack that we could use?

Thanks again,and appreciate your help.

AloshkaD, 2016-06-25:
Thanks Jose,
That hasn't worked neither. Still getting the same error. I originally changed the installation directory from bitbucket to github, so I didn't have an issue there. Kindly note that the Human Machine Interface directory doesn't exist any more and I had it forked before  it  was deleted by the author.

AloshkaD, 2016-06-25:
Jose, thank you.  
Unfortunately that didn't work, here is the error message

"
----------------------------
Installing Mavros dependencies
----------------------------
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros--control-toolbox
!!!Error installing Mavros dependencies. Distro non supported!!!
------------------------------------------------------
Installing Ueyecamera drivers
------------------------------------------------------
./installation/installation_dep_script.sh: line 83: cd: /installation/drivers: No such file or directory
sudo: ./ueyesdk-setup-4.60-usb-amd64.gz.run: command not found
-------------------------------------------------------
Fixing CMakeLists.txt to be able to open QTCreator
-------------------------------------------------------
rm: cannot remove ‘CMakeLists.txt’: No such file or directory
cp: cannot stat ‘/opt/ros/jade/share/catkin/cmake/toplevel.cmake’: No such file or directory
-------------------------------------------------------
Compiling the Aerostack
-------------------------------------------------------
-Setting ROS workspace
/home/deleteme/workspace/ros/quadrotor_stack_catkin_ws/src/quadrotor_stack/setup.sh: line 5: /devel/setup.bash: No such file or directory
-Setting roscore in localhost
/home/deleteme/temp/installation_script.sh: line 118: catkin_make: command not found
deleteme@ubuntu:~$ 

"

AloshkaD, 2016-06-26:
Thanks Jose,

I tried again but I've got the error below. I also tried to create a new vertual machine with a fresly installed ROS but that did'nt work either.

"
-- Performing Test COMPILER_SUPPORTS_STD_CXX11
-- Performing Test COMPILER_SUPPORTS_STD_CXX11 - Success
CMake Error at quadrotor_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  quadrotor_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/CMakeLists.txt:16 (include)

-- Configuring incomplete, errors occurred!
See also "/home/test/workspace/ros/quadrotor_stack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/test/workspace/ros/quadrotor_stack_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

"

hridaybavle, 2016-06-27:
What version of ROS are you using, is it ROS indigo or jade ?

joselusl, 2016-06-27:
I have fixed the installer (https://github.com/Vision4UAV/Aerostack_installer ) that had a bug.
I am still working in fixing the dependencies installation script (https://github.com/Vision4UAV/Aerostack/blob/master/installation/installation_dep_script.sh) that has some bugs also. I need some other maintainers to fix their packages to have the dependencies correct. I will push the update as soon as the other packages are fixed.
Sorry for the inconveniences.

AloshkaD, 2016-06-28:
Great, let me know if I can do anything to help. I appreciate your hard work and support.

joselusl, 2016-06-29:
Hi,
I have already pushed an update to Aerostack.
Please, try a fresh installation using the Aerostack installer: https://github.com/Vision4UAV/Aerostack_installer as it is specified in the Aerostack Wiki (https://github.com/Vision4UAV/Aerostack/wiki/Getting-Started ).
Everything should work now (at least it is working on my computer). If you find any other issue, let me know and I will try to fix it as soon as possible.
Bests

Nazul, 2016-06-29:
I'm seeing a similar error when trying to install. Today I started the Aerostack install process as specified by the Wiki article. It failed. I re-downloaded the installer script, then run it again and I found the same issue.

Results from the second time I ran the script.

It starts, then it shows an error but continues working:

```
WARNING: Package name "droneSpeedPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
CMake Error at aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/CMakeLists.txt:16 (include)

-- Configuring incomplete, errors occurred!
See also "/home/marioc/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/marioc/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
-------------------------------------------------------
Sourcing the ROS Aerostack WS
-------------------------------------------------------
-------------------------------------------------------
Fetching the required git submodule
-------------------------------------------------------
-------------------------------------------------------
Installing dependencies
-------------------------------------------------------
-----------------------
Installing ncurses
-----------------------
```

Then, it continues with other dependencies until it fails as follows (and stops):

```
WARNING: Package name "droneSpeedPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
CMake Error at aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/CMakeLists.txt:16 (include)

-- Configuring incomplete, errors occurred!
See also "/home/marioc/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/marioc/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

Any ideas if this is related or not? Or if it's an issue with my environment.

AloshkaD, 2016-06-30:
Unfortunately Jose, that hasn't fixed  it for me too. I it will help, I can either give you remote access to my machine or send you my virtual drive OS.

joselusl, 2016-06-30:
Hi guys!
Thank you for your patience and sorry for the inconveniences.
(I think) I have already fixed it. There was a problem in the installer with the environmental variables $AEROSTACK_STACK and $AEROSTACK_WORKSPACE that were not exported to the script that was installing the dependencies, so the installation was failing and therefore, when compiling it was giving errors.
I tested it on a fresh VM with just Ubuntu 14.04 and ROS Jade.
Please, do a fresh installation of the Aerostack using the installer, and the problems should be solved.
I keep the issue open until you confirmed that you didn't have any problem!
Bests

joselusl, 2016-06-30:
Guys, there is still a bug. Give me some more time!
Sorry :(

joselusl, 2016-06-30:
ok, now it worked!
Please do a fresh installation (the installer has been updated).
Enjoy!
Sorry again for all the inconveniences!

AloshkaD, 2016-06-30:
Jose, I'm sorry that you have to go through this :) I've got the same error! This time the installer continued  installing the packages and in the end it gave this error

/usr/bin/ld: cannot find -llibmikrokopter
collect2: error: ld returned 1 exit status
make[2]: *** [/home/test/workspace/ros/aerostack_catkin_ws/devel/lib/mikrokopter_driver/mikrokopter_driver_node] Error 1
make[1]: *** [aerostack_stack/stack/hardware_interface/drivers_platforms/driver_mikrokopter_flight_controller/driver_mikrokopter_flight_controller/CMakeFiles/mikrokopter_driver_node.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

[image]

joselusl, 2016-07-01:
Damn!
I thought I was fixed this issue in this commit: https://github.com/Vision4UAV/Aerostack/commit/f77cba75b3d24d639c8816dd7957b0592c94932c
Apparently, I still need to work on it. 
Any case, good news is that we advanced and mavlink is not giving any error anymore.
Could you pease do me two favors?
1. Could you check the file $AEROSTACK_STACK/stack/hardware_interface/drivers_platforms/driver_mikrokopter_flight_controller/CMakeLists.txt, line 76? You should have this: add_dependencies(mikrokopter_driver_node libmikrokopter)
2. Could you run catkin_make again: cd $AEROSTACK_WORKSPACE; source $AEROSTACK_STACK/setup.sh; catkin_make

AloshkaD, 2016-07-01:
Yes absolutely, I will head to the Lab in a few hours and check those.

I think the Mavlink gave an error but despite that it advanced with the installation. I will run it again and crosscheck.

Nazul, 2016-07-01:
I tried again, and I have a different error. Last error is the same as AloshkaD reported, but the previous messages are different.

```
-- Build files have been written to: /home/marioc/workspace/ros/aerostack_catkin_ws/build
####
#### Running command: "make -j1 -l1" in "/home/marioc/workspace/ros/aerostack_catkin_ws/build"
####
Scanning dependencies of target _catkin_empty_exported_target
[  0%] Built target _catkin_empty_exported_target
Scanning dependencies of target levmar
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/lm.c.o
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/Axb.c.o
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/misc.c.o
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/lmlec.c.o
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/lmbc.c.o
[  0%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/lmblec.c.o
[  1%] Building C object aerostack_stack/stack/libraries/lib_levmar-2.6/CMakeFiles/levmar.dir/lmbleic.c.o
Linking C shared library /home/marioc/workspace/ros/aerostack_catkin_ws/devel/lib/liblevmar.so
[  1%] Built target levmar
Scanning dependencies of target aruco_lib
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/ar_omp.cpp.o
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/arucofidmarkers.cpp.o
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/board.cpp.o
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/boarddetector.cpp.o
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/cameraparameters.cpp.o
[  1%] Building CXX object aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/chromaticmask.cpp.o
In file included from /home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:29:0:
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/include/aruco_lib/chromaticmask.h:61:5: error: ‘EM’ in namespace ‘cv’ does not name a type
     cv::EM _classifier;
     ^
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp: In constructor ‘EMClassifier::EMClassifier(unsigned int)’:
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:43:5: error: ‘_classifier’ was not declared in this scope
     _classifier = cv::EM(2, cv::EM::COV_MAT_DIAGONAL, cv::TermCriteria(cv::TermCriteria::COUNT , 4, FLT_EPSILON));
     ^
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:43:19: error: ‘EM’ is not a member of ‘cv’
     _classifier = cv::EM(2, cv::EM::COV_MAT_DIAGONAL, cv::TermCriteria(cv::TermCriteria::COUNT , 4, FLT_EPSILON));
                   ^
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:43:19: note: suggested alternative:
In file included from /usr/local/include/opencv2/ml/ml.hpp:48:0,
                 from /home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/include/aruco_lib/chromaticmask.h:33,
                 from /home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:29:
/usr/local/include/opencv2/ml.hpp:732:20: note:   ‘cv::ml::EM’
 class CV_EXPORTS_W EM : public StatModel
                    ^
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:43:33: error: ‘cv::EM’ has not been declared
     _classifier = cv::EM(2, cv::EM::COV_MAT_DIAGONAL, cv::TermCriteria(cv::TermCriteria::COUNT , 4, FLT_EPSILON));
                                 ^
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp: In member function ‘void EMClassifier::train()’:
/home/marioc/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/src/source/chromaticmask.cpp:103:5: error: ‘_classifier’ was not declared in this scope
     _classifier.train(samples);
     ^
make[2]: *** [aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/src/source/chromaticmask.cpp.o] Error 1
make[1]: *** [aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_lib/CMakeFiles/aruco_lib.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
```

If I run `cd $AEROSTACK_WORKSPACE; source $AEROSTACK_STACK/setup.sh; catkin_make`, I see basically the same error.

I checked `$AEROSTACK_STACK/stack/hardware_interface/drivers_platforms/driver_mikrokopter_flight_controller/CMakeLists.txt` and line 76 is commented. I removed the # character and it's running so far.

AloshkaD, 2016-07-02:
The good news is Mavlink is compiling without any issues. 

Here is what I have in my Cmakelists.txt, obviously I don't have that line.

[image]

AloshkaD, 2016-07-02:
I added this line "add_dependencies(mikrokopter_driver_node libmikrokopter)" to the cmakelists. I went to source the stack but I got "source $AEROSTACK_STACK/setup.sh
bash: /setup.sh: No such file or directory"

I have the workspace and aerostack_catkin_ws directories.

AloshkaD, 2016-07-04:
Ok guys, i think I had it fixed. see the fig and let me know  if that's showing succesfull compilation of the packages
[image]

To fix the issues I added the line to my cmakelist as Jose recommended. then I went to aerostack_catkin_ws and invoked catkin_make . it didn't report any errors. Then I ran ~/temp/installation_script.sh

joselusl, 2016-07-04:
Hi,
Great you finally managed to install it properly.
Probably the issue you all were having when doing the "fresh installation" I was recommending to do.
My idea was that you manually remove both the aerostack workspace folder and the aerostack installer, then download the aerostack installer again (because I did some updates) and then run the installation command again.
Sorry I was not clear enough.
I am going to improve the aerostack installer to autonomously do this.
Bests

closed 2016-07-04 by joselusl

AloshkaD, 2016-07-04:
Thanks Jose,
Actually, I was removing the old installation and the folder all together and gitting the new installer.Perhaps Catkin was not being invoked, somehow. Anyways, I'm glad it's fixed and I can share my virtual drive with whoever needs it

joselusl, 2016-07-07:
Thank you. I am currently uploading a Virtual Machine that is going to be accessible from the Wiki

AloshkaD, 2016-07-07:
Thanks Jose,
In my research I use docker, which is something you may consider as well.

https://www.docker.com/

joselusl, 2016-07-10:
Thank you, we will take a look!