cvar-upm/aerostack #5: Missing droneMsgsROS header files (AliveSignal.h, droneManagerStatus.h)

AloshkaD, 2016-06-25:
There seem to be missing header files in droneMsgsROS, the compilation terminates with

 "Invoking "make -j4 -l4" failed"

Here is the output
"
ation_console.cpp.o
[ 81%] Built target lib_cvglogger
[ 81%] Built target droneArucoEye
[ 81%] Built target droneEKFStateEstimator
[ 81%] Built target droneVisualMarkersEyeSimulator
[ 81%] Built target driver_px4flow_interface_ROSModule
[ 82%] Built target droneBrainROSModule
[ 82%] Built target droneCV2DTo3DROSModule
[ 82%] Built target droneIARC14BrainROSModule_lib
[ 82%] Built target droneIARC14BrainROSModule_with_trajplanner_lib
[ 83%] Built target droneIARC14MissionSchedulerROSModule
[ 83%] Built target droneMidLevelAutopilotROSModule
[ 83%] Built target droneMidLevelAutopilotROSModule_joy_converter_to_midlevelautopilot_cmd
[ 83%] Built target droneMissionPlannerROSModule_lib
[ 84%] Built target droneObstacleProcessorVisualMarksROSModule
[ 84%] Built target droneSimulatoriRobotCV_lib
[ 84%] Built target multicameraRobotTrackerSimulator_lib
[ 84%] Built target droneTrajectoryPlannerROSModule
[ 84%] Built target testing_obstacles
[ 84%] Building CXX object quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/src/controller/source/telemetry_state_receiver.cpp.o
In file included from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/vehicle_view.h:19:0,
                 from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/vehicle_view.cpp:12:
/home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/telemetry_state_receiver.h:24:38: fatal error: droneMsgsROS/AliveSignal.h: No such file or directory
 #include "droneMsgsROS/AliveSignal.h"
                                      ^
compilation terminated.
In file included from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/main.cpp:15:0:
/home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/main_window.h:29:45: fatal error: droneMsgsROS/droneManagerStatus.h: No such file or directory
 #include <droneMsgsROS/droneManagerStatus.h>
                                             ^
compilation terminated.
In file included from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/communication_console.h:14:0,
                 from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/communication_console.cpp:11:
/home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/telemetry_state_receiver.h:24:38: fatal error: droneMsgsROS/AliveSignal.h: No such file or directory
 #include "droneMsgsROS/AliveSignal.h"
                                      ^
compilation terminated.
In file included from /home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/telemetry_state_receiver.cpp:18:0:
/home/alex/workspace/ros/quadrotor_stack_catkin/src/quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/src/controller/source/../include/telemetry_state_receiver.h:24:38: fatal error: droneMsgsROS/AliveSignal.h: No such file or directory
 #include "droneMsgsROS/AliveSignal.h"
                                      ^
compilation terminated.
make[2]: *** [quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/src/controller/source/main.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 84%] Built target testing_societyPose
[ 84%] Built target droneVisualMarkersEyeSimulatorROSModule
make[2]: *** [quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/src/controller/source/vehicle_view.cpp.o] Error 1
[ 84%] Built target droneVisualMarkersLocalizerROSModule_lib
[ 85%] Built target droneYawCommanderROSModule_lib
[ 85%] Built target lib_cvgloggerROS
[ 85%] Built target batteryPelicanROSModule
make[2]: *** [quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/src/controller/source/communication_console.cpp.o] Error 1
[ 85%] Built target droneCommandPelicanROSModule
[ 85%] Built target rotationAnglesPelicanROSModule
[ 85%] [ 85%] Built target batteryOktoROSModule
Built target droneCommandOktoROSModule
[ 85%] Built target rotationAnglesOktoROSModule
[ 85%] Built target droneArucoEyeROSModule_lib
[ 85%] Built target droneComputerVisionIARC14GroundRobots
[ 86%] Built target droneComputerVisionIARC14Keypoints
[ 86%] Built target droneInterfaceROSModule
[ 86%] Built target droneInterface_jp_ROSModule
[ 87%] Built target droneLoggerROSModule
[ 87%] Built target dronePelicanLoggerROSModuleOneImage
[ 87%] Built target dronePelicanLoggerROSModule
[ 87%] [ 87%] Built target batteryARDroneROSModule
Built target altitudeARDroneROSModule
[ 87%] Built target ardrone_driver
[ 87%] Built target droneCommandARDroneROSModule
[ 87%] Built target bottomCameraARDroneROSModule
[ 87%] Built target droneStatusARDroneROSModule
[ 87%] Built target frontCameraARDroneROSModule
[ 87%] Built target groundSpeedARDroneROSModule
[ 87%] Built target imuARDroneROSModule
[ 87%] Built target magnetometerARDroneROSModule
[ 87%] Built target pressureARDroneROSModule
[ 87%] Built target rotationAnglesARDroneROSModule
[ 87%] [ 87%] Built target droneArchitectureRvizInterfaceROSModuleNode
Built target temperatureARDroneROSModule
[ 87%] Built target videoPublisherNode
[ 87%] [ 87%] Built target lib_cvgutils_code_tests2
Built target lib_cvgutils_code_tests
[ 87%] Built target arucoEyeTest
[ 87%] Built target droneTrajectoryPlanner2dTest
[ 88%] Built target droneSimulator
[ 88%] Built target lib_cvglogger_code_tests
[ 88%] Built target driver_camera_ueye
[ 88%] Built target dronePBVSPositionMidLevelController
[ 89%] Built target droneSpeedPositionMidLevelController
[ 89%] Built target px4flowAltitudeROSModule
[ 89%] Built target px4flowInterfaceROSModule
[ 89%] [ 89%] Built target droneBrainROSModuleNode
Built target droneCVGroundRobots2DTo3D
[ 89%] Built target droneCVKeypoints2DTo3D
[ 89%] Built target droneIARC14BrainROSModule_with_trajplanner
[ 89%] Built target droneMidLevelAutopilotROSModuleNode
[ 90%] Built target droneIARC14BrainROSModule
[ 90%] Built target droneMissionPlannerROSModule
[ 90%] Built target joy_converter_to_midlevelautopilot_cmd
[ 91%] Built target droneObstacleProcessor2dVisualMarksROSModule
[ 91%] [ 91%] Built target dronePelicanSimulatorROSModule
Built target joy_converter_to_pelican_cmd_lib
[ 92%] Built target droneSimulatorGridROSModule
[ 92%] Built target droneSimulatorROSModule
[ 92%] [ 92%] Built target droneSimulatoriRobotCV
Built target multicameraRobotTrackerSimulator
[ 92%] Built target droneTrajectoryPlanner2dROSModule
[ 92%] [ 93%] Built target droneVisualMarkersLocalizerROSModule
Built target droneVisualMarkersEyeSimulatorROSModuleNode
[ 93%] Built target droneYawCommanderROSModule
[ 93%] [ 93%] Built target lib_cvgloggerROS_code_tests
Built target lib_cvgloggerROS_code_tests2
[ 93%] Built target lib_cvgloggerROS_code_tests3
[ 93%] Built target lib_cvgloggerROS_code_tests5
[ 93%] Built target lib_cvgloggerROS_code_tests4
[ 93%] Built target lib_cvgloggerROS_code_tests6
[ 93%] Built target droneOktoSimulatorROSModule
[ 93%] Built target droneEKFStateEstimatorROSModule
make[2]: *** [quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/src/controller/source/telemetry_state_receiver.cpp.o] Error 1
make[1]: *** [quadrotor_stack/stack/HMI/hmi_cvg_stack/human_machine_interface/CMakeFiles/human_machine_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Built target joy_converter_to_okto_cmd_lib
[ 93%] [ 93%] Built target driver_camera_ueye_ROSModule
Built target droneArucoEyeROSModule
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
"

joselusl, 2016-06-27:
Hi,
I can see both messages in the package droneMsgsROS:
AliveSignal: https://bitbucket.org/joselusl/dronemsgsros/src/51ce2b07d3bf5e0d1e3a464af19332365fd3f855/msg/AliveSignal.msg?at=master&fileviewer=file-view-default
droneManagerStatus: https://bitbucket.org/joselusl/dronemsgsros/src/51ce2b07d3bf5e0d1e3a464af19332365fd3f855/msg/droneManagerStatus.msg?at=master&fileviewer=file-view-default
Please, check that you have downloaded the correct version of Aerostack. 
If you can also see them, the problem might be related to this open issue: https://github.com/Vision4UAV/Aerostack/issues/4
I am currently working on it. Sorry for the inconveniences.
Bests

AloshkaD, 2016-06-28:
Thanks Jose, Yes I have the right package. I have the ".msg" too but the problem is with the header file ".h". There are many header files available but many else are missing.

joselusl, 2016-06-30:
I am pretty sure this issue was related to https://github.com/Vision4UAV/Aerostack/issues/4 
I am going to close it, and keep the other one open.

closed 2016-06-30 by joselusl