cvar-upm/aerostack #7: AscTec Pelican launchers

CV-Ursula, 2016-07-04:
Hello, 
I have been going thorough the documentation and noticed there are no intructions on how to run the framework on the AscTec Pelican. 
Could you please provide some instructions on how to run the framework on the Pelican or maybe even a basic script?

joselusl, 2016-07-04:
Hello,
Thank you for relying on Aerostack. We have worked many times with the AscTec Pelican, but as you remark, we didn't have documented it yet on the Wiki. We will try to do as soon as we can.
In the meantime, the git submodules $AEROSTACK_STACK/configs (https://bitbucket.org/joselusl/quadrotor_stack_configs) and $AEROSTACK_STACK/launchers (https://bitbucket.org/Vision4UAV/aerostack_launchers) have other branches with more config files and scripts for launching different missions and drones.
Feel free to go to these submodules, git checkout other branches and explore them by your own.
Bests

CV-Ursula, 2016-07-04:
Thank you, 
will take a look at the them.

closed 2016-08-08 by joselusl