cvar-upm/aerostack #8: DRONE_STACK  is not set

AloshkaD, 2016-07-08:
When I run "${IBVS_STACK}/launchers/parrot_IBVSController_launcher_Release.sh " I get the error below. Also, the  DRONE_STACK folder does not exist. I need to use is for adding the config file to "\Config " directory for the camera. 

Invalid <arg> tag: environment variable 'DRONE_STACK' is not set. 

Arg xml is <arg default="$(env DRONE_STACK)" name="my_stack_directory"/>
The traceback for the exception was written to the log file

[image]

joselusl, 2016-07-10:
Hi,
In the previous versions of Aerostack (V0.x), we used two environmental variables: $DRONE_STACK and $DRONE_WORKSPACE.
Now, in the current version (V1.x), we changed these variables to: $AEROSTACK_STACK and $AEROSTACK_WORKSPACE.
We tried to update all the launchers, but we might have missed some.
Please, if you ever see one of the old environmental variables, change them to the new ones. Everything should work.
Bests

AloshkaD, 2016-07-11:
Thank you, that worked for me!

closed 2016-07-11 by AloshkaD