cvar-upm/aerostack #9: Error ehne trying to launch Pelican Launchers

CV-Ursula, 2016-08-29:
Hello, 
I have tried to run the Pelican launchers provided in #7. 

However when I try to launch the scripts in the order specidfied in the pelican_real_flight_instructions
file I get the error which can be seen on the screenshot.

[image]

joselusl, 2016-09-20:
First of all, I guess the node that is giving you errors is the “autopilot_node” from the package “asctec_autopilot”. Could you please confirm me this?
This node is the one used as a hardware interface with the AscTec Pelican.

Second of all, as I said in this issue (https://github.com/Vision4UAV/Aerostack/issues/3 ), we didn’t developed this package (asctec_autopilot, included in http://wiki.ros.org/asctec_drivers ), we just used it and created a common interface for Aerostack (this common interface is the requirement to be able to use all the drones in a similar way). This interface is the package “driverPelicanROSModule” (https://bitbucket.org/ramon_suarez_fernandez/driverpelicanrosmodule/ ). 
As the asctec_drivers metapackage (https://github.com/ccny-ros-pkg/asctec_drivers , that includes the asctec_autopilot) is superold, when we catkinized the Aerostack, we forked it and catkinized the minimum packages we needed to fly the Asctec pelican (https://bitbucket.org/joselusl/asctec_drivers/ ).
In addition, asctec_drivers depended on mav_tools (http://wiki.ros.org/mav_tools and https://github.com/ccny-ros-pkg/mav_tools ) that is also superold and we catkinized (https://bitbucket.org/joselusl/mav_tools/ ), again only the basic packages.
I personally didn’t worked directly with this. Ramon, in cc’s, did, so he might give you more details. But, my suggestion is to try to read all the documentation and the how-to of the original packages so you can understand them properly, together with the HL/LL flashing/connection requirements.

Third, after telling you that Aerostack is using the asctec_drivers metapackage, I must say that there is a newer (and better) metapackage for managing the AscTec autopilot, the “asctec_mav_framework” (http://wiki.ros.org/asctec_mav_framework ). The migration is in our TO DO list, but we never had the time to do it.
The migration should be simple (in terms of coding) and we only need to develop an interface between this package and Aerostack equivalent to “driverPelicanROSModule”.
The reason why we have not done it is because we are not using the pelican anymore (only for minor tests) and developing a package is not only coding it, but also testing it with real flights.
If you feel like you want to do it and contribute to Aerostack, I would be very happy to guide you (but again, the coding might be super simple, and the big difficulty is the testing).

suarez-ramon, 2016-09-20:
Hello, This error usually appears when the connector is not correctly used between the asctec atomboard or mastermind to the LL of the Autopilot. Can you check these connections correctly and the ports if you are using mastermind or atomboard.

CV-Ursula, 2016-09-27:
Hello, 
@joselusl Thank you for the detailed description of the framework. I have read the documentation on the links you have sent. Now everything is working fine. 

@suarez-ramon you were right. The connections were not according to thoes specified on the AscTec website. I have connected them properly and now it works. Thank you for your help.

closed 2016-09-27 by CV-Ursula