cvar-upm/aerostack #11: Aerostack con Ubuntu Xenial y ROS Kinetic

jbermejoalonso, 2017-03-23:
Hola,

Estoy intentando instalar Aerostack en un Mac, con una máquina virtual de VirtualBox sobre Ubuntu 16.04.2 (Xenial) y ROS Kinetic. Durante el proceso se han producido varios mensajes:

1- advertencia sobre nombres de los paquetes estilo:  WARNING. PAckage name "droneSpeechROSModule" does not follow the naming convention... Y así con varios más.

2- error sobre audio: audio_capture: No definition of [gstreamer0.10-plugins-ugy] for OS version [xenial], sound_play: No definition of [python-gst] for OS version [xenial]

3- Pero sobre todo un error de configuración
[image]

Un compañero utilizando también Xenial y ROS Kinetic ha obtenido un  mensaje similar de error de configuración. Así que tengo la duda de si Aerostack funciona con Xenial y ROS Kinetic. Alguna sugerencia?

Gracias

Julia

pascualcampoy, 2017-03-23:
La que envia este correo es Julia Bermejo, que trabaja con Ricardo Sanz
y que ahora están intentando utilizar el Aerostack en su proyecto con Ixion

Ambos en copia

por favor tratadlo con especial interés para resolverlo

gracias

Pascual Campoy

Full Professor U.P.M.
Head of Computer Vision & Aerial Robotics
Center of Automatics and Robotics
www.vision4uav.eu <http://www.vision4uav.eu/>
e-mail: pascual.campoy@upm.es
telf.:     +34 91336 3061

Visiting Prof. at TUDelft
             p.campoy@tudelft.nl

joselusl, 2017-03-23:
Hola Julia,

Te contesto rapidamente:
- No te preocupes por los warnings. Los conocemos. Estamos trabajando en eliminarlos, pero no afectan al funcionamiento, solo te llenan la pantalla..
- Esta versión no es compatible con Ubuntu Xenial y ROS Kinetic... lo más seguro es que los errores que aparecen se deben a esta incomparibilidad.. tenemos ya una versión experimental compatible, pero todavía no la hemos volcado en el repositorio público (ya que seguimos testeandola). Te recomiendo que uses ROS Jade y Ubuntu 14.04. Si realmente necesitas Kinetic, hazmelo saber por correo y te doy acceso a la version experimental.

closed 2017-03-23 by joselusl

jbermejoalonso, 2017-03-23:
Hola,

Entendido. Gracias por la pronta resolución.

Lo intentaremos con Jade. Sino, ya os digo.

Julia