cvar-upm/aerostack #14: Cannot find lib_cvgutil-config.cmake

virenbajaj, 2017-06-14:
### This is the error I get while running the installation script with ubuntu 16.04 and ROS kinetic:

```CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "lib_cvgutils" with
  any of the following names:

    lib_cvgutilsConfig.cmake
    lib_cvgutils-config.cmake

  Add the installation prefix of "lib_cvgutils" to CMAKE_PREFIX_PATH or set
  "lib_cvgutils_DIR" to a directory containing one of the above files.  If
  "lib_cvgutils" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  aerostack_stack/stack_deprecated/simulation_system/droneVisualMarkersEyeSimulator/droneVisualMarkersEyeSimulator/CMakeLists.txt:47 (find_package)

-- Could not find the required component 'lib_cvgutils'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "lib_cvgutils" with
  any of the following names:

    lib_cvgutilsConfig.cmake
    lib_cvgutils-config.cmake

  Add the installation prefix of "lib_cvgutils" to CMAKE_PREFIX_PATH or set
  "lib_cvgutils_DIR" to a directory containing one of the above files.  If
  "lib_cvgutils" provides a separate development package or SDK, be sure it
  has been installed.
```
How can I fix this??

virenbajaj, 2017-06-14:
More specifically `$AEROSTACK_STACK/stack/libraries/lib_cvgutils/` is an empty directory. Is this where the cmake file should have been?

joselusl, 2017-06-17:
It seems that this git submodule has not been properly downloaded when you run the installation script.
Could you please run this command: "cd $AEROSTACK_STACK && git submodule update --init"? With this command you will download all the git submodules.
If there was an issue during the installation, you might need to run again the following command before "catkin_make" it: "$AEROSTACK_STACK/installation/installation_dep_script.sh". With this command you will install all the package dependencies.
Please remember to source to ROS (source /opt/ros/$ROS_DISTRO/setup.sh).

virenbajaj, 2017-06-19:
stack/common/drone_process cannot have https://bitbucket.org/DavidNPalacios/drone_process.git/
cloned into it because its a dead link. So "git submodule update --init" doesn't finish execution. 

While installing package dependencies, I still get the following error:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
droneModuleInterfaceROS: Cannot locate rosdep definition for [lib_cvgutils]
droneVisualMarkersEyeSimulator: Cannot locate rosdep definition for [lib_cvgutils]
```

joselusl, 2017-06-19:
You are right! We recently did a change on the drone_process package (and some others) that seems to create a dead link! [This is the new repo for drone_process: https://bitbucket.org/albertcp_/drone_process ]
We are going to try to solve this issue, making it working again as soon as possible.
I am very sorry for all the inconveniences...

virenbajaj, 2017-06-20:
Ok, thanks for doing it quickly!

joselusl, 2017-06-20:
It should be fixed now! There was a pull-request that was solving this issue and that we forgot to merge..

virenbajaj, 2017-06-20:
Awesome thats fixed now, but I think a similar issue causes this new error:
```
CMake Error at aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/CMakeLists.txt:16 (include)

-- Configuring incomplete, errors occurred!
```

hridaybavle, 2017-06-22:
Hi, 
Mavlink ros doesn't seem to be installed in your system. 
Try 
sudo apt-get install ros-kinetic-mavlink

virenbajaj, 2017-06-23:
Ok that fixed it. :+1: 
git submodule update --init --recursive completes. 
but  ~/temp/installation_script.sh gives a new error: 
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
QWT_INCLUDE_DIR
   used as include directory in directory /home/viren/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/ground_control_system/human_machine_interface/human_machine_interface_process
QWT_LIBRARY
    linked by target "human_machine_interface" in directory /home/viren/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/ground_control_system/human_machine_interface/human_machine_interface_process

-- Configuring incomplete, errors occurred!
See also "/home/viren/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/viren/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

and installation_dep_script executes with error: 
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
audio_capture: No definition of [gstreamer0.10-plugins-ugly] for OS version [xenial]
sound_play: No definition of [python-gst] for OS version [xenial]
audio_play: No definition of [gstreamer0.10-plugins-ugly] for OS version [xenial]
```

martin-molina, 2017-06-23:
This problem seems to be related to Ubuntu 16.04. The current version of the human_machine_interface was tested on Ubuntu 14.04.

I would recommend to use now Ubuntu 14.04 and ROS Jade. In a few weeks (a the end of July) we plan to release a new version of Aerostack with new features for Ubuntu 16.04 and ROS Kinetic.

joselusl, 2017-06-26:
If you are in a rush, you need to use Ubuntu 16.04, and you can survive without using the human machine interface, you can put a CATKIN_IGNORE file in this package, and everything should compile properly (not guaranteed, but worked for me).

virenbajaj, 2017-06-27:
OK so I switched to Ubuntu 14.04 and ROS Jade and after a few quick fixes im stuck with:
```
-- +++ processing catkin package: 'tld_tracker'
-- ==> add_subdirectory(aerostack_stack/stack/feature_extraction_system/opentld/ros_opentld/tld_tracker)
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Could NOT find Subversion (missing:  Subversion_SVN_EXECUTABLE) 
CMake Error at /usr/share/cmake-2.8/Modules/ExternalProject.cmake:1241 (message):
  error: could not find svn for checkout of opentld
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/ExternalProject.cmake:1805 (_ep_add_download_command)
  aerostack_stack/stack/feature_extraction_system/opentld/ros_opentld/tld_tracker/CMakeLists.txt:26 (externalproject_add)

-- Configuring incomplete, errors occurred!
See also "/home/viren/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/viren/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

joselusl, 2017-06-27:
> -- Could NOT find Subversion (missing:  Subversion_SVN_EXECUTABLE) 
It seems that you don't have subversion installed. Try to install it (sudo apt-get install subversion)

JacekCieslak95, 2017-10-18:
It seems, that I've got the same problem, which @virenbajaj  had on the beginning.
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
droneModuleInterfaceROS: Cannot locate rosdep definition for [lib_cvgutils]
droneVisualMarkersEyeSimulator: Cannot locate rosdep definition for [lib_cvgutils]`

when I use command "git submodule update --init" (given by @joselusl ) I receive information:
`Cloning into 'stack/common/drone_process'...
remote: Not Found
fatal: repository 'https://bitbucket.org/albertcp_/drone_process.git/' not found
Clone of 'https://bitbucket.org/albertcp_/drone_process.git' into submodule path 'stack/common/drone_process' failed
`
Could you please tell me, how to solve it?
(I'm running Ubuntu 14.04, ROS Jade)

martin-molina, 2017-10-18:
Thank you for your message. There was an error caused by maintenance work. We have corrected the error, so it should work now.
Please, let us know if you have any problem.
Thanks.

JacekCieslak95, 2017-10-18:
Everything is working perfectly! Thank you for such fast response.

closed 2017-11-10 by martin-molina