cvar-upm/aerostack #15: Error al isntalar Aerostack

patrilopez, 2017-06-22:


martin-molina, 2017-06-23:
Parece ser que estás utilizando una versión de Qwt desactualizada. Prueba por ejemplo:

sudo apt-get install libqwt6 libqwt-dev

patrilopez, 2017-06-26:
Muchas gracias! He conseguido instalarlo haciendo lo que me has comentado.
Aunque, ahora al intentar lanzar Aerostack me da error. Siguiendo la guía escribo el siguiente comando en el terminal: ./ardrone_basic.sh
Al hacerlo se abren nuevas ventanas de terminal y en una de ellas aparece el siguiente mensaje de error:
process[drone1/human_machine_interface_process-1]: started with pid [6685]
*** Error in `/home/usuario/workspace/ros/aerostack_catkin_ws/devel/lib/human_machine_interface/human_machine_interface': realloc(): invalid pointer: 0x000000000092fda0 ***
[drone1/human_machine_interface_process-1] process has died [pid 6685, exit code -6, cmd /home/usuario/workspace/ros/aerostack_catkin_ws/devel/lib/human_machine_interface/human_machine_interface __name:=human_machine_interface_process __log:=/home/usuario/.ros/log/522f914e-5a51-11e7-993e-00224dabcf6a/drone1-human_machine_interface_process-1.log].
log file: /home/usuario/.ros/log/522f914e-5a51-11e7-993e-00224dabcf6a/drone1-human_machine_interface_process-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

¿Cómo podría solucionarlo?
Gracias de antemano!

martin-molina, 2017-06-28:
Vuelve a ejecutar el script de instalación de dependencias tal como se indica en la guía de instalación:

$AEROSTACK_STACK/installation/installation_dep_script.sh

Para revisar la correcta instalación de las distintas versiones de Qt y Opencv puedes ejecutar el siguiente comando:

sudo apt-get install libopencv-dev opencv-data ros-jade-opencv-apps libqt4-dev libqt4-core

Después deberías intentar volver a compilar y comprobar si funciona. 

Avísanos por favor con el resultado.

closed 2017-11-10 by martin-molina