cvar-upm/aerostack #19: Aerostack 2.0 - instalation problem

JacekCieslak95, 2017-10-27:
I was trying to install Aerostack 2.0, but during instalation there was an error:
```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "pugixml" with any
  of the following names:

    pugixmlConfig.cmake
    pugixml-config.cmake

  Add the installation prefix of "pugixml" to CMAKE_PREFIX_PATH or set
  "pugixml_DIR" to a directory containing one of the above files.  If
  "pugixml" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_eye_core/CMakeLists.txt:51 (find_package)

-- Could not find the required component 'pugixml'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "pugixml" with any
  of the following names:

    pugixmlConfig.cmake
    pugixml-config.cmake

  Add the installation prefix of "pugixml" to CMAKE_PREFIX_PATH or set
  "pugixml_DIR" to a directory containing one of the above files.  If
  "pugixml" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_eye_core/CMakeLists.txt:51 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jacek/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jacek/workspace/ros/aerostack_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

I had similar issue during installation of Aerostack 1.0, so I decided to use command:
`git submodule update --init`
I was asked to give username and password to bickbucket.org. I did it, but there was an error (I'm sure, that username and password are correct. I tried also with my email and password with the same results)

```
Cloning into 'stack/hardware_interface/drivers_platforms/driver_dji_matrice/dji_sdk_ros'...
Username for 'https://bitbucket.org': Jacek_Cieslak 
Password for 'https://Jacek_Cieslak@bitbucket.org': 
remote: Invalid username or password. If you log in via a third party service you must ensure you have an account password set in your account profile.
fatal: Authentication failed for 'https://bitbucket.org/ramon_suarez_fernandez/dji_sdk_ros.git/'
fatal: clone of 'https://bitbucket.org/ramon_suarez_fernandez/dji_sdk_ros.git' into submodule path 'stack/hardware_interface/drivers_platforms/driver_dji_matrice/dji_sdk_ros' failed
```
Could you please help me to solve this problem?

martin-molina, 2017-10-27:
Thank you for your message. We have corrected the problem related to permissions. So, it should work now.

JacekCieslak95, 2017-10-27:
I tried again, and I can not see any change. Problem is still the same.
Even, if I try to open this link in my browser
**https://bitbucket.org/ramon_suarez_fernandez/dji_sdk_ros.git**
I have error: Access denied

Lowyiiii, 2017-10-28:
@martin-molina I have same problem
 fatal: Authentication failed for 'https://bitbucket.org/ramon_suarez_fernandez/driver_rotors_simulator_process.git/'
fatal: clone of 'https://bitbucket.org/ramon_suarez_fernandez/driver_rotors_simulator_process.git' into submodule path 'stack/hardware_interface/drivers_platforms/driver_rotors_simulator/driver_rotors_simulator_process' failed

martin-molina, 2017-10-31:
We have corrected the permissions already. Please let us know if you have problems now.

JacekCieslak95, 2017-10-31:
Installations works fine right now. Thank you!
I'm having some problems with behavior tree, but I think they're connected with topic remapping, which I made to connect Aerostack to Gazebo.

albertcp, 2017-11-02:
Hi @JacekCieslak95,

It seems the argument written for the behavior GO_TO_POINT is incorrect. This behavior only accepts the following arguments: "coordinates: [x, y, z]" and "relative_coordinates: [x, y, z]".

I have updated the [page that explains behaviors](https://github.com/Vision4UAV/Aerostack/wiki/Catalog-of-Behaviors) because the arguments were written in uppercase and with parenthesis, which is incorrect.

I hope this fixes the issue.

soheil2017, 2017-11-10:
I have a problem in installation. after the running **~/temp/installation_script.sh** , I got following error

collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_eye_ros/CMakeFiles/aruco_eye_ros.dir/build.make:241: recipe for target '/home/soheil/workspace/ros/aerostack_catkin_ws/devel/lib/libaruco_eye_ros.so' failed
make[2]: *** [/home/soheil/workspace/ros/aerostack_catkin_ws/devel/lib/libaruco_eye_ros.so] Error 1
make[2]: *** Deleting file '/home/soheil/workspace/ros/aerostack_catkin_ws/devel/lib/libaruco_eye_ros.so'
CMakeFiles/Makefile2:46401: recipe for target 'aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_eye_ros/CMakeFiles/aruco_eye_ros.dir/all' failed
make[1]: *** [aerostack_stack/stack/feature_extraction_system/aruco_eye/aruco_eye/aruco_eye_ros/CMakeFiles/aruco_eye_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 60%] Building CXX object aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/mavros/CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o
[ 60%] Building CXX object aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/mavros/CMakeFiles/mavros.dir/src/lib/rosconsole_bridge.cpp.o
[ 60%] Linking CXX shared library /home/soheil/workspace/ros/aerostack_catkin_ws/devel/lib/libmavros.so
[ 60%] Built target mavros
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

Could you please help me to solve this problem?

hridaybavle, 2017-11-14:
Hi Soheil, 

The problem looks like you have an internal compiler error, which generally occurs when using arm architecture systems like odroid or rasberry pi. Can you please mention which computer are you using to build the Aerostack?

If its an odroid  or rasberry pi try the following steps:

1. cd  $AERSTACK_WORKSPACE
2. catkin_make -j1 -l1 (this step will take around 30 mins to compile but you should not get this error),

Thanks, 
Regards.

soheil2017, 2017-11-20:
Hi friend. Thank you. 
my problem solved.

closed 2017-11-20 by martin-molina