cvar-upm/aerostack #21: Multiple ar.drones in rviz / preparing own behaviors

pkaczmarek95, 2017-11-08:
Hello, 

I've got two questions about Aerostack. I would like to know if there is any way to launch a rviz simulation with two or more ar.drones? The other one is about behaviors - is it possible to write own ones?

martin-molina, 2017-11-08:
Hi, we have added some pages explaining how to write your own behaviors:

https://github.com/Vision4UAV/Aerostack/wiki/How-to-Program-a-New-Behavior

martin-molina, 2017-11-13:
Hi,
This is the configuration what we have used to run Aerostack + Gazebo
CPU: Intel Core i7 (7th Gen)
GPU: Nvidia GEFORCE GTX 1050 Ti
RAM: 16 GB

We have used also Intel Core i5 and 8GB RAM using Ruiz.

martin-molina, 2017-11-13:
Hi, 
I confirm you that it is possible to launch a RViz simulation with several drones. We are preparing a script to help you with that. We'll let you know when it is ready.

albertcp, 2017-11-15:
Hi @JacekCieslak95 

I tried to create a new behavior following the tutorial (in order to generate the error you describe) and everything worked fine. For that reason, I have updated the [tutorial](https://github.com/Vision4UAV/Aerostack/wiki/How-to-Program-a-New-Behavior) adding an example of the ```package.xml``` file and the ```CMakeLists.xml``` file.

I hope this helps you.

PS: The step 5 is also updated. It describes that in order to activate the behavior, you just need to launch the behavior created and Aerostack. Then, by calling a ROS service inside the terminal, you can activate the behavior created.

JacekCieslak95, 2017-11-16:
Everything works fine right now, but I have one more question: I can use my behavior without any problem with python missions, but is it plausible to use it with behavior tree? (my behavior is not shown on list of available behaviors). Should I edit any file?

martin-molina, 2017-11-16:
Thank you for your message. You are right in your comment, so we are going to correct it. We expect to have it in less than two weeks. We will inform you when this change is done.

martin-molina, 2018-01-10:
This message is to confirm that the change was already done.

closed 2018-01-10 by martin-molina