cvar-upm/aerostack #22: Problema QWT_LIB en Ubuntu 14.04 LT con ROS jade

santofimia, 2017-11-22:
Estoy teniendo problemas al instalar aerostack en Ubuntu 14.04 LT con ROS jade, siguiendo los pasos que se indican en la web: 

git clone https://github.com/Vision4UAV/Aerostack_installer ~/temp
~/temp/installation_script.sh

Me da un error con la libreria QWT:

[image]

Y comprobando el script de instalación, hay un punto he dice lo siguiente:

[image]

Donde dice que los paquetes libqwt-headers y libqwt-qt5-dev no los puede encontrar.

Y buscando he encontrado que estos paquetes no estan en la distribucion de ubunut 14.04 LTS:
https://packages.ubuntu.com/search?keywords=qwt

y no se si el error se debe puramente a esto, y el script no esta preparado para esta distribucón o es otra cosa externa. En todo caso, ¿sabe como podría solucionarlo?

Muchas Gracias,
Pablo Santofimia Ruiz

martin-molina, 2017-11-23:
Hola Pablo,

La versión actual de Aerostack es 2.0 (distribución Levant) y necesita Ubuntu 16.04 como soporte a su nuevo interfaz gráfico.

Para resolver el problema que comentas puedes hacer tres cosas:
1. Pasarte a Ubuntu 16.04 (recomendado)
2. Descargar la versión anterior de Aerostack (versión 1.0 distribución Gregale) que opera en Ubuntu 14.04.
3. Usar Aerostack 2.0 sin el interfaz de usuario gráfico.

La versión anterior Aerostack 1.0 la puedes descargar del siguiente enlace:

https://github.com/Vision4UAV/Aerostack/releases/tag/v1-gregale

Y su documentación en la página:

https://github.com/Vision4UAV/Aerostack/wiki/List-of-distributions

Espero que con ello lo puedas resolver. Avísanos para cualquier problema.

santofimia, 2017-11-23:
Vale perfecto, 
me pasare a ubuntu 16.04 si es el que tiene soporte.
Muchas Gracias!!

closed 2017-12-28 by martin-molina