cvar-upm/aerostack #23: OpenTLD GUI

hoffa95, 2017-11-27:
Hi,

I am trying to figure out which launch file should I run, if I want to use the OpenTLD GUI and the visual servoing processes. Just like in your cvg_ardrone2_ibvs package. I didn't find any tutorials about this in your Wiki.
It would be nice if you could help me.

Thanks.

jespestana, 2017-11-27:
Hi hoffa95,

My colleagues from the UPM would be able to tell you the exact launching procedures. Otherwise, my guess is once you download the full stack, you should have in the "launchers" folder the repository:
https://bitbucket.org/Vision4UAV/aerostack_launchers
In this repository there are several launchers for the AR Drone: 
https://bitbucket.org/Vision4UAV/aerostack_launchers/src/835c635c89dfe323f1980d05ebc54af4f87c2613/ardrone_launch/?at=master
I  think that it should work similarly to the cvg_ardrone2_ibvs package once you get it running.

hoffa95, 2017-11-27:
Thanks for your fast reply! 
My problem solved!

closed 2017-11-27 by hoffa95

mark-91, 2022-01-09:
> Hi hoffa95,
>
> My colleagues from the UPM would be able to tell you the exact launching procedures. Otherwise, my guess is once you download the full stack, you should have in the "launchers" folder [...]

i try to run the code but still do not work 
i see the link and download but i do not know Which script should i run first 
can you please help?