cvar-upm/aerostack #24: How to use "FOLLOW_OBJECT_IMAGE" behavior?

pkaczmarek95, 2017-12-15:
Hello,

I am trying to write a python mission that includes "FOLLOW_OBJECT_IMAGE" behavior. The description of it says that "the operator selects the image to follow through the user interface" - however in the "camera views" bookmark I am not able to do it in any way. How to use that behavior?

Thanks for considering my request.

albertcp, 2017-12-19:
Hi!

This page of the [wiki](https://github.com/Vision4UAV/Aerostack/wiki/Catalog-of-Processes) has every process that need to be launched in order to use the behavior FOLLOW_OBJECT_IMAGE. Check if every of the processes mentioned are launched.
I see that the required launch files are not uploaded. I will write you as soon as I upload both files.

Regarding the other question, inside the launchers directory, you can find a script called "rvizInterface_Test.sh". You should run as many Aerostacks as drones you need and then you can visualise them by executing the script mentioned.

I hope you solve all your problems.

JacekCieslak95, 2017-12-20:
I am trying to launch simulation script (simulated_quadrotor_basic.sh) with input argument, to launch many Aerostacks. I saw, that 

> NUMID_DRONE=$1
so first given argument should be an drone number. Running command 
./simulated_quadrotor_basic.sh 2
to give this drone NUMID_DRONE=2 gives strange result: Aerostack can't see roscore / master:
> roscore/master is not yet running, will wait for it to start

pkaczmarek95, 2017-12-21:
Thanks, @albertcp , I have updated Aerostack and I have required files now. I was trying to run my mission mentioned in post #1. Following advices on the page 'processes' I launched:
droneTrackerEyeROSModule.launch
DroneIBVSControllerROSModule.launch
droneOpenTLDTranslatorROSModule.launch
and 
opentld_for_IBVSController.launch
opentld_gui_for_IBVSController.launch
After launching all of this (and my node that publish data from a webcam to /drone1/camera/front/image_raw) I had this nodes active:
[image]
and new window TLD Gui appeared. However it was empty. 
[image]
Mission still doesn't even appear in execution viewer during a mission. How to use it properly?

albertcp, 2017-12-22:
@JacekCieslak95 You can type ```./simulated_quadrotor_basic.sh 2 <roscore_ip>``` for now.

The problems you had seems to be related to the roscore IP. When you assign only the droneID it also changes the roscore IP. I'll write you as soon as the script is fixed.

albertcp, 2017-12-22:
@pkaczmarek95 Which drone id are you using to launch Aerostack? 

Every Aerostack process should have the same drone_id to work properly. I can see from your ```rosnode list``` output that the processes you are launching have drone0 assigned, but I don't know the drone_id assigned to the Aerostack processes.

To launch any Aerostack process as droneX you can do 2 different things (Being X: 0,1,2,3,4...):
1. Add the process inside the launcher if they are not already. To do so, copy and paste one and change the process name and the .launch file.
2. When you launch the process via terminal add this parameter at the end of each process: ```drone_id_namespace:=droneX```. DroneX should be the same as the rest of Aersotack's processes. For example, with the behavior GO_TO_POINT the command has to be: ```roslaunch behavior_go_to_point behavior_go_to_point.launch drone_id_namespace:=drone1```

If this is the problem, the camera should be appeared when the behavior FOLLOW_OBJECT_IMAGE is activated.

pkaczmarek95, 2017-12-22:
@albertcp 
Thanks, you were right, I modified launch files and change drone_id from 0 to 1 (as 1 is default drone_id for Aerostack). Now, after running all processes and aerostack my rosnode list looks like this:
https://pastebin.com/FGvTMXVH
There is some progress because FOLLOW_OBJECT_IMAGE finally appears in an execution viewer when i run a mission. However TLD Gui is still empty. Should I run something more?

albertcp, 2017-12-30:
It should appear a freeze image of the drone the moment every process starts.

Is the TLD GUI still empty after executing the behavior FOLLOW_OBJECT_IMAGE?. 

Anyway, I'll make some tests to see if no image is received by the TLD GUI.

closed 2018-10-11 by martin-molina