cvar-upm/aerostack #25: Checking battery level in a mission written in python

pkaczmarek95, 2018-01-05:
Hello,

I'd like to know if there is any way to check the battery level during execution of mission. On your wiki I read that there is a belief "charge(battery, x)" but every time I try to use:
 success, unification = api.consultBelief('charge(self, ?X)')
I get the message that query was unsuccessful. I guesAm I using it wrong way? Or is it one of these beliefs that will be implemented in the future?

martin-molina, 2018-01-07:
In your question you say that you tried:
success, unification = api.consultBelief('charge(self, ?X)')
But the belief expression should be charge(battery,?X) instead of charge(self,?X).
Please, let us know if this solves the problem.

martin-molina, 2018-01-07:
We have now this predicate for battery:
battery_level(self,?X)
The values of X are qualitative.

pkaczmarek95, 2018-01-13:
@martin-molina  Thanks for reply, it works fine now.

closed 2018-02-16 by martin-molina