cvar-upm/aerostack #26: battery 0%

soheil2017, 2018-02-04:
Hi every one

for the first time I got following problem that i couldn't find any solution for it.  my AR-Drone 2.0 is connected to AeroStack with ubuntu 16.04 (ROS-kinetic) but the battery is showing 0%. but indeed the battery is full charge. also I don't have camera image. in addition I separately  tested it by ardron-autonoumy package and it works perfectly.  I re-started the Ar Drone but Aerostack  doesn't work yet. do you know the problem?

albertcp, 2018-02-06:
Hi @soheil2017 ,

Did you start Aerostack after being connected to the AR-Drone 2.0?

The steps to initiate Aerostack correctly are:
1. Connect to the ARDrone wifi
2. Ejecute ./ardrone_basic.sh

If you follow these steps and still doesn't work, verify that the launcher is using the correct IP address of the ARDrone.

soheil2017, 2018-02-06:
dear @albertcp 
I am exactly doing these steps and I also verified  IP address of the ARDrone but it doesn't work yet. I am sure the connection of AR Drone is correct because I tested it with other packages such as ardrone_autonoumy and ORB_SLAM and PTAM.  

[image]

Lowyiiii, 2018-02-06:
Hi every one
I have the same problem

albertcp, 2018-02-08:
Can you check if the drone_id value is the same for all Aerostack processes?

You can do that by typing ```rosnode list``` and also ```rostopic list``` and paste the results here.

soheil2017, 2018-02-08:
@albertcp 
 this is my rostopic list:

soheil@soheil-X555LNB:~$ rostopic list 
/common_beliefs
/drone1/ArucoSlam_EstimatedPose
/drone1/ArucoSlam_EstimatedPose_v2
/drone1/ArucoSlam_EstimatedSpeed
/drone1/ArucoSlam_EstimatedSpeeds
/drone1/ArucoSlam_LandarmkList
/drone1/EstimatedPose_droneGMR_wrt_GFF
/drone1/EstimatedPose_iRobotGMR_wrt_droneGMR
/drone1/EstimatedSpeed_droneGMR_wrt_GFF
/drone1/SOEstimatedPose
/drone1/SOEstimatedSpeeds
/drone1/all_beliefs
/drone1/altitude
/drone1/approved_action
/drone1/ardrone/bottom/camera_info
/drone1/ardrone/bottom/image_raw
/drone1/ardrone/bottom/image_raw/compressed
/drone1/ardrone/bottom/image_raw/compressed/parameter_descriptions
/drone1/ardrone/bottom/image_raw/compressed/parameter_updates
/drone1/ardrone/bottom/image_raw/compressedDepth
/drone1/ardrone/bottom/image_raw/compressedDepth/parameter_descriptions
/drone1/ardrone/bottom/image_raw/compressedDepth/parameter_updates
/drone1/ardrone/bottom/image_raw/theora
/drone1/ardrone/bottom/image_raw/theora/parameter_descriptions
/drone1/ardrone/bottom/image_raw/theora/parameter_updates
/drone1/ardrone/camera_info
/drone1/ardrone/front/camera_info
/drone1/ardrone/front/image_color
/drone1/ardrone/front/image_color/compressed
/drone1/ardrone/front/image_color/compressed/parameter_descriptions
/drone1/ardrone/front/image_color/compressed/parameter_updates
/drone1/ardrone/front/image_color/compressedDepth
/drone1/ardrone/front/image_color/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_color/compressedDepth/parameter_updates
/drone1/ardrone/front/image_color/theora
/drone1/ardrone/front/image_color/theora/parameter_descriptions
/drone1/ardrone/front/image_color/theora/parameter_updates
/drone1/ardrone/front/image_mono
/drone1/ardrone/front/image_mono/compressed
/drone1/ardrone/front/image_mono/compressed/parameter_descriptions
/drone1/ardrone/front/image_mono/compressed/parameter_updates
/drone1/ardrone/front/image_mono/compressedDepth
/drone1/ardrone/front/image_mono/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_mono/compressedDepth/parameter_updates
/drone1/ardrone/front/image_mono/theora
/drone1/ardrone/front/image_mono/theora/parameter_descriptions
/drone1/ardrone/front/image_mono/theora/parameter_updates
/drone1/ardrone/front/image_proc_debayer/parameter_descriptions
/drone1/ardrone/front/image_proc_debayer/parameter_updates
/drone1/ardrone/front/image_proc_rectify_color/parameter_descriptions
/drone1/ardrone/front/image_proc_rectify_color/parameter_updates
/drone1/ardrone/front/image_proc_rectify_mono/parameter_descriptions
/drone1/ardrone/front/image_proc_rectify_mono/parameter_updates
/drone1/ardrone/front/image_raw
/drone1/ardrone/front/image_raw/compressed
/drone1/ardrone/front/image_raw/compressed/parameter_descriptions
/drone1/ardrone/front/image_raw/compressed/parameter_updates
/drone1/ardrone/front/image_raw/compressedDepth
/drone1/ardrone/front/image_raw/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_raw/compressedDepth/parameter_updates
/drone1/ardrone/front/image_raw/theora
/drone1/ardrone/front/image_raw/theora/parameter_descriptions
/drone1/ardrone/front/image_raw/theora/parameter_updates
/drone1/ardrone/front/image_rect
/drone1/ardrone/front/image_rect/compressed
/drone1/ardrone/front/image_rect/compressed/parameter_descriptions
[... 143 lines cut ...]
/drone1/droneCommand
/drone1/droneCommunicationManager
/drone1/droneGroundSpeed
/drone1/droneInterface
/drone1/droneLocalizer
/drone1/droneObstacleDistanceCalculator
/drone1/droneObstacleProcessor
/drone1/droneOdometryStateEstimator
/drone1/droneRotationAngles
/drone1/droneStatusARDroneROSModule
/drone1/droneTrajectoryController
/drone1/droneTrajectoryPlanner
/drone1/droneYawPlanner
/drone1/frontCameraARDroneROSModule
/drone1/graphical_user_interface_process
/drone1/imuARDroneROSModule
/drone1/magnetometerARDroneROSModule
/drone1/pressureARDroneROSModule
/drone1/process_monitor_process_main
/drone1/python_based_mission_interpreter_process
/drone1/resource_manager_process
/drone1/self_localization_selector_process
/drone1/task_based_mission_planner_process
/drone1/temperatureARDroneROSModule
/rosout

soheil2017, 2018-02-08:
my rostopic list : 

soheil@soheil-X555LNB:~$ rostopic list 
/common_beliefs
/drone1/ArucoSlam_EstimatedPose
/drone1/ArucoSlam_EstimatedPose_v2
/drone1/ArucoSlam_EstimatedSpeed
/drone1/ArucoSlam_EstimatedSpeeds
/drone1/ArucoSlam_LandarmkList
/drone1/EstimatedPose_droneGMR_wrt_GFF
/drone1/EstimatedPose_iRobotGMR_wrt_droneGMR
/drone1/EstimatedSpeed_droneGMR_wrt_GFF
/drone1/SOEstimatedPose
/drone1/SOEstimatedSpeeds
/drone1/all_beliefs
/drone1/altitude
/drone1/approved_action
/drone1/ardrone/bottom/camera_info
/drone1/ardrone/bottom/image_raw
/drone1/ardrone/bottom/image_raw/compressed
/drone1/ardrone/bottom/image_raw/compressed/parameter_descriptions
/drone1/ardrone/bottom/image_raw/compressed/parameter_updates
/drone1/ardrone/bottom/image_raw/compressedDepth
/drone1/ardrone/bottom/image_raw/compressedDepth/parameter_descriptions
/drone1/ardrone/bottom/image_raw/compressedDepth/parameter_updates
/drone1/ardrone/bottom/image_raw/theora
/drone1/ardrone/bottom/image_raw/theora/parameter_descriptions
/drone1/ardrone/bottom/image_raw/theora/parameter_updates
/drone1/ardrone/camera_info
/drone1/ardrone/front/camera_info
/drone1/ardrone/front/image_color
/drone1/ardrone/front/image_color/compressed
/drone1/ardrone/front/image_color/compressed/parameter_descriptions
/drone1/ardrone/front/image_color/compressed/parameter_updates
/drone1/ardrone/front/image_color/compressedDepth
/drone1/ardrone/front/image_color/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_color/compressedDepth/parameter_updates
/drone1/ardrone/front/image_color/theora
/drone1/ardrone/front/image_color/theora/parameter_descriptions
/drone1/ardrone/front/image_color/theora/parameter_updates
/drone1/ardrone/front/image_mono
/drone1/ardrone/front/image_mono/compressed
/drone1/ardrone/front/image_mono/compressed/parameter_descriptions
/drone1/ardrone/front/image_mono/compressed/parameter_updates
/drone1/ardrone/front/image_mono/compressedDepth
/drone1/ardrone/front/image_mono/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_mono/compressedDepth/parameter_updates
/drone1/ardrone/front/image_mono/theora
/drone1/ardrone/front/image_mono/theora/parameter_descriptions
/drone1/ardrone/front/image_mono/theora/parameter_updates
/drone1/ardrone/front/image_proc_debayer/parameter_descriptions
/drone1/ardrone/front/image_proc_debayer/parameter_updates
/drone1/ardrone/front/image_proc_rectify_color/parameter_descriptions
/drone1/ardrone/front/image_proc_rectify_color/parameter_updates
/drone1/ardrone/front/image_proc_rectify_mono/parameter_descriptions
/drone1/ardrone/front/image_proc_rectify_mono/parameter_updates
/drone1/ardrone/front/image_raw
/drone1/ardrone/front/image_raw/compressed
/drone1/ardrone/front/image_raw/compressed/parameter_descriptions
/drone1/ardrone/front/image_raw/compressed/parameter_updates
/drone1/ardrone/front/image_raw/compressedDepth
/drone1/ardrone/front/image_raw/compressedDepth/parameter_descriptions
/drone1/ardrone/front/image_raw/compressedDepth/parameter_updates
/drone1/ardrone/front/image_raw/theora
/drone1/ardrone/front/image_raw/theora/parameter_descriptions
/drone1/ardrone/front/image_raw/theora/parameter_updates
/drone1/ardrone/front/image_rect
/drone1/ardrone/front/image_rect/compressed
/drone1/ardrone/front/image_rect/compressed/parameter_descriptions
/drone1/ardrone/front/image_rect/compressed/parameter_updates
[... 142 lines cut ...]
/drone1/droneCommand
/drone1/droneCommunicationManager
/drone1/droneGroundSpeed
/drone1/droneInterface
/drone1/droneLocalizer
/drone1/droneObstacleDistanceCalculator
/drone1/droneObstacleProcessor
/drone1/droneOdometryStateEstimator
/drone1/droneRotationAngles
/drone1/droneStatusARDroneROSModule
/drone1/droneTrajectoryController
/drone1/droneTrajectoryPlanner
/drone1/droneYawPlanner
/drone1/frontCameraARDroneROSModule
/drone1/graphical_user_interface_process
/drone1/imuARDroneROSModule
/drone1/magnetometerARDroneROSModule
/drone1/pressureARDroneROSModule
/drone1/process_monitor_process_main
/drone1/python_based_mission_interpreter_process
/drone1/resource_manager_process
/drone1/self_localization_selector_process
/drone1/task_based_mission_planner_process
/drone1/temperatureARDroneROSModule
/rosout

albertcp, 2018-02-09:
It seems every process is launched correctly and it should connect correctly. Can you check the terminals and see if there are processes that crashed or give some error?

Lowyiiii, 2018-02-09:
hello @albertcp 
this is my errors
i have the same problem in ardrone and bebop2. also I have found a link that seems the problem is because of drivers
http://wiki.ros.org/kinetic/Migration
========================

[drone1/imuARDroneROSModule-1] process has died [pid 7674, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/imuARDroneROSModule __name:=imuARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-imuARDroneROSModule-1.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-imuARDroneROSModule-1*.log
[drone1/temperatureARDroneROSModule-2] process has died [pid 7687, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/temperatureARDroneROSModule __name:=temperatureARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-temperatureARDroneROSModule-2.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-temperatureARDroneROSModule-2*.log
[drone1/magnetometerARDroneROSModule-3] process has died [pid 7713, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/magnetometerARDroneROSModule __name:=magnetometerARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-magnetometerARDroneROSModule-3.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-magnetometerARDroneROSModule-3*.log
[drone1/droneBattery-4] process has died [pid 7729, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/batteryARDroneROSModule __name:=droneBattery __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneBattery-4.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneBattery-4*.log
[drone1/droneAltitude-5] process has died [pid 7747, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/altitudeARDroneROSModule __name:=droneAltitude __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneAltitude-5.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneAltitude-5*.log
[drone1/droneRotationAngles-6] process has died [pid 7763, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/rotationAnglesARDroneROSModule __name:=droneRotationAngles __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneRotationAngles-6.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneRotationAngles-6*.log
[drone1/droneGroundSpeed-7] process has died [pid 7787, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/groundSpeedARDroneROSModule __name:=droneGroundSpeed __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneGroundSpeed-7.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneGroundSpeed-7*.log
[drone1/pressureARDroneROSModule-8] process has died [pid 7804, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/pressureARDroneROSModule __name:=pressureARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-pressureARDroneROSModule-8.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-pressureARDroneROSModule-8*.log
[drone1/frontCameraARDroneROSModule-11] process has died [pid 7849, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/frontCameraARDroneROSModule __name:=frontCameraARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-frontCameraARDroneROSModule-11.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-frontCameraARDroneROSModule-11*.log
[drone1/droneCommand-12] process has died [pid 7855, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/droneCommandARDroneROSModule __name:=droneCommand __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneCommand-12.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneCommand-12*.log
[drone1/droneStatusARDroneROSModule-9] process has died [pid 7827, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/droneStatusARDroneROSModule __name:=droneStatusARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneStatusARDroneROSModule-9.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-droneStatusARDroneROSModule-9*.log
[drone1/bottomCameraARDroneROSModule-10] process has died [pid 7834, exit code -6, cmd /home/lowyi/workspace/ros/aerostack_catkin_ws/devel/lib/driverParrotARDroneROSModule/bottomCameraARDroneROSModule __name:=bottomCameraARDroneROSModule __log:=/home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-bottomCameraARDroneROSModule-10.log].
log file: /home/lowyi/.ros/log/99aafe02-0d96-11e8-b4ab-6817298b2a6d/drone1-bottomCameraARDroneROSModule-10*.log

soheil2017, 2018-02-16:
@albertcp 
thanks for realizing the new version of AeroStack. in new version, the problem of my battery is solved.

closed 2018-02-16 by martin-molina