cvar-upm/aerostack #28: SLAM_dunk

soheil2017, 2018-02-25:
Hi guys 

I am currently using Bebop2 and I wanna use AeroStack package. since the using SLAM_dunk  in this package is necessary, but I can not find SLAM_dunk  to buy. also I asked from parrot company and they said me that we don't sell it anymore. do you know how can I get it and where? in addition, is it possible to use AeroStack for Bebop2  with some modifications? 

Thanks

martin-molina, 2018-02-26:
I confirm that Aerostack requires SLAM Dunk for Bebop 2 in order to have the adequate frequencies. If SLAM Dunk is not available now, perhaps you can use AR Drone 2.

martin-molina, 2019-03-03:
The new version of Aerostack (3.0) works with Bebop2 without slam dunk.

closed 2019-03-03 by martin-molina