cvar-upm/aerostack #36: unknown type

QiuMingZS, 2019-03-24:
When I launched the simulation, I can not get the image from the camera, and find:
 
* /drone1/camera/bottom/image_raw [unknown type]
* /drone1/camera/front/image_raw [unknown type]

What's the reason?

The complete code is as follows:
`guanzheng@guanzheng-CN15S:~$ rosnode info /drone1/graphical_user_interface_process
--------------------------------------------------------------------------------
Node [/drone1/graphical_user_interface_process]
Publications: 
 * /drone1/command/dAltitude [droneMsgsROS/droneDAltitudeCmd]
 * /drone1/command/dYaw [droneMsgsROS/droneDYawCmd]
 * /drone1/command/pitch_roll [droneMsgsROS/dronePitchRollCmd]
 * /drone1/droneControllerYawRefCommand [droneMsgsROS/droneYawRefCommand]
 * /drone1/droneMissionPlannerCommand [droneMsgsROS/droneMissionPlannerCommand]
 * /drone1/dronePositionRefs [droneMsgsROS/dronePositionRefCommandStamped]
 * /drone1/droneSpeedsRefs [droneMsgsROS/droneSpeeds]
 * /drone1/requested_action [droneMsgsROS/actionData]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /drone1/EstimatedPose_droneGMR_wrt_GFF [droneMsgsROS/dronePose]
 * /drone1/all_beliefs [droneMsgsROS/AllBeliefs]
 * /drone1/altitude [droneMsgsROS/droneAltitude]
 * /drone1/approved_action [unknown type]
 * /drone1/battery [droneMsgsROS/battery]
 * /drone1/behavior_event [droneMsgsROS/BehaviorEvent]
 * /drone1/camera/bottom/image_raw [unknown type]
 * /drone1/camera/front/image_raw [unknown type]
 * /drone1/camera/overlay/image_raw/big [sensor_msgs/Image]
 * /drone1/camera/overlay/image_raw/small [sensor_msgs/Image]
 * /drone1/command/dAltitude [droneMsgsROS/droneDAltitudeCmd]
 * /drone1/command/dYaw [droneMsgsROS/droneDYawCmd]
 * /drone1/command/high_level [droneMsgsROS/droneCommand]
 * /drone1/command/pitch_roll [droneMsgsROS/dronePitchRollCmd]
 * /drone1/completed_action [unknown type]
 * /drone1/completed_mission [droneMsgsROS/CompletedMission]
 * /drone1/current_task [unknown type]
 * /drone1/droneManagerStatus [unknown type]
 * /drone1/droneMissionHLCommandAck [unknown type]
 * /drone1/droneSpeedsRefs [droneMsgsROS/droneSpeeds]
 * /drone1/droneTrajectoryAbsRefCommand [droneMsgsROS/dronePositionTrajectoryRefCommand]
 * /drone1/estimated_pose [droneMsgsROS/dronePose]
 * /drone1/estimated_speeds [unknown type]
 * /drone1/event [unknown type]
 * /drone1/ground_speed [droneMsgsROS/vector2Stamped]
 * /drone1/imu [sensor_msgs/Imu]
 * /drone1/list_of_active_behaviors [droneMsgsROS/ListOfBehaviors]
 * /drone1/magnetometer [geometry_msgs/Vector3Stamped]
 * /drone1/obstacles [droneMsgsROS/obstaclesTwoDim]
 * /drone1/pressure [sensor_msgs/FluidPressure]
 * /drone1/process_error_unified_notification [droneMsgsROS/ProcessError]
 * /drone1/processes_performance [droneMsgsROS/ProcessDescriptorList]
 * /drone1/rotation_angles [geometry_msgs/Vector3Stamped]
 * /drone1/skills_list [unknown type]
 * /drone1/societyPose [droneMsgsROS/societyPose]
 * /drone1/status [droneMsgsROS/droneStatus]
 * /drone1/surface_inspection_image [unknown type]
 * /drone1/temperature [sensor_msgs/Temperature]
 * /drone1/trajectoryControllerPositionReferencesRebroadcast [droneMsgsROS/dronePose]
 * /drone1/trajectoryControllerSpeedReferencesRebroadcast [droneMsgsROS/droneSpeeds]
 * /drone1/wifiIsOk [std_msgs/Bool]
 * /public_event [unknown type]

Services: 
 * /drone1/graphical_user_interface_process/get_loggers
 * /drone1/graphical_user_interface_process/set_logger_level

[... 223 lines cut ...]
 * topic: /drone1/list_of_active_behaviors
    * to: /drone1/behavior_coordinator_process (http://guanzheng-CN15S:45774/)
    * direction: inbound
    * transport: TCPROS
 * topic: /drone1/all_beliefs
    * to: /drone1/belief_manager_process (http://guanzheng-CN15S:36049/)
    * direction: inbound
    * transport: TCPROS
 * topic: /drone1/behavior_event
    * to: /drone1/behavior_self_localize_by_odometry (http://guanzheng-CN15S:38824/)
    * direction: inbound
    * transport: TCPROS
 * topic: /drone1/behavior_event
    * to: /drone1/behavior_take_off (http://guanzheng-CN15S:35132/)
    * direction: inbound
    * transport: TCPROS
 * topic: /drone1/behavior_event
    * to: /drone1/behavior_go_to_point (http://guanzheng-CN15S:36197/)
    * direction: inbound
    * transport: TCPROS
 * topic: /drone1/behavior_event
    * to: /drone1/behavior_land (http://guanzheng-CN15S:38670/)
    * direction: inbound
    * transport: TCPROS
`

martin-molina, 2019-03-25:
If you are using droneSimulator, you'll have to consider that it does not generate images from camera.

QiuMingZS, 2019-03-25:
Can I use the camera in the simulation?

QiuMingZS, 2019-03-25:
Is there any other way to simulate using the camera?

closed 2019-03-26 by QiuMingZS