cvar-upm/aerostack #37: Issue with vrep and aerostack

jimis093, 2019-04-03:
Hey your project is really nice 
Can you help me to find a way to setting up the vrep with aerostack ? Thanx !

martin-molina, 2019-04-03:
In the next version of Aerostack (version 3.0) we plan to include an example describing how to use Aerostack with V-Rep. We expect to release version 3.0 at the end of May or the beginning of June. Thanks.

jimis093, 2019-04-03:
Thanx for response !!! i want to ask you if i can with this version to use the vrep ! Thanx

Στάλθηκε από το Ταχυδρομείο Yahoo σε Android

martin-molina, 2019-04-03:
Yes, but you have to create the interfaces.

jimis093, 2019-04-03:
Thanx for helping !!!!!!

Στάλθηκε από το Ταχυδρομείο Yahoo σε Android

jimis093, 2019-04-03:
The interfaves of ros or the vrep ? 

Στάλθηκε από το Ταχυδρομείο Yahoo σε Android 
 
  Στις Πέμ, 4 Απρ, 2019 στις 1:38 π.μ., ο χρήστηςdimitris dimitris<gdimitris.connex@yahoo.gr> έγραψε:   Thanx for helping !!!!!!

Στάλθηκε από το Ταχυδρομείο Yahoo σε Android

jimis093, 2019-04-04:
Plz can you tell me a few things about this ? I need for exam work ...

martin-molina, 2019-04-05:
You can use inter-process communication provided by ROS (e.g., ROS topics) to communicate VRep and Aerostack.

jimis093, 2019-04-06:
hello finaly i made it to connect the vrep with aerostack , i want to ask you how my drone will follows as path some images ?  thanx :)

martin-molina, 2019-04-08:
Could you please give us more details to clarify your question "how my drone will follows as path some images ?". Thank you.

jimis093, 2019-04-08:
hello , i have this  refinery and i want my drone to follow the pipes . can you tell me how i can do this ? thanx

jimis093, 2019-04-08:
also i have an issue, when the drone is in trajectrory mode , then it can not move correct, but when i control it from keyboard it moves perfect .. what can i do for that ? thanx

jimis093, 2019-04-09:
Hello again , i have some questions , First of all what  can i do with exporting files from aerostack ( i mean how i can use the exporting files , yaml and python files in a project from vrep ) ?
Sec , the drone of vrep and the 'drone  visual maker of aerostack ' they are confused and i get something like that (img 111111111111111111111111).

And with what way can i have α resault like this https://www.youtube.com/watch?v=0Z3Qgj1BhnM ? 

thanx .. Sory for i beeing annoying but is to immportant for me ! thnx

jimis093, 2019-04-09:
Hello again , i have some questions , First of all what  can i do with exporting files from aerostack ( i mean how i can use the exporting files , yaml and python files in a project from vrep ) ?
Sec , the drone of vrep and the 'drone  visual maker of aerostack ' they are confused and i get this (img 111111111111111111111111).

And with what way can i have α resault like this https://www.youtube.com/watch?v=0Z3Qgj1BhnM ? 

thanx .. Sory for i beeing annoying but is to immportant for me ! thnx

martin-molina, 2019-04-09:
Hi, it is difficult for us to help you now with this issue because we don't have an implemented reusable solution yet. As I said, we are working now to have it ready for the next release (in May or June). However, if we have it before, we will let you know. Thank you.

closed 2019-07-26 by martin-molina