cvar-upm/aerostack #38: Mutil-UAVs cooperation

MOMOHTT, 2019-04-18:
Dear professor：
   I want to achieve Muti-UAVs cooperation using Aerosatck，but I don’t know how to run mutil-UAV simulation programs in a GUI, and how to achieve inter-communication between UAVs.

martin-molina, 2019-04-22:
If you want to use a simulator (e.g., Gazebo or VREP) with different UAVs, you need to program new interfaces between the simulator and Aerostack (through ROS). Unfortunately, we don't provide these interfaces with Aerostack 2.1, but we are working to include them in future versions.

MOMOHTT, 2019-04-22:
Thank you so much for your reply！And I'm looking forward to the Aerosatck3.0.
签名由网易邮箱大师定制

MOMOHTT, 2019-05-07:
Dear professor:
    I have two questions. first，I have read some papers published by your team. The experimental part is done indoors. If I want to use outdoor scenes, do I have to implement an outdoor positioning functional component (such as GPS) by myself, and then add it to the launcher file? Second，Can multiple UAVs be placed in Aerostack for simulation? If can, how to do this？

martin-molina, 2019-05-14:
> do I have to implement an outdoor positioning functional component (such as GPS) by myself, and then add it to the launcher file? 
Yes, this is the approach.
> Can multiple UAVs be placed in Aerostack for simulation?
You have to create an instance of Aerostack for each UAV.

closed 2019-07-26 by martin-molina