cvar-upm/aerostack #39: simulation using rotors

QiuMingZS, 2019-04-22:
Hi, in the stack, I find the part of the rotors_simulator, and try to use it. But I find that in the file  README,  the ros version is indigo, but the Aerostack is corresponded to kinetic,.
[image]

hridaybavle, 2019-04-22:
Hi, 

The aerostack will install the proper rotors version for you when you install the aerostack. You will have it installed properly if you have already installed Aerostack. You dont need to follow the rotors instructions separately.

QiuMingZS, 2019-04-22:
thanks😁 王冠政 邮箱：guanzhengnpu@gmail.com 签名由 网易邮箱大师 定制 On 04/22/2019 22:13, Hriday Bavle wrote: Hi, The aerostack will install the proper rotors version for you when you install the aerostack. You will have it installed properly if you have already installed Aerostack. You dont need to follow the rotors instructions separately.

QiuMingZS, 2019-04-25:
Thank you so much

closed 2019-04-25 by QiuMingZS