cvar-upm/aerostack #44: Problema en la instalación de Aerostack 3.0 en Ubuntu 18.04 con ROS Melodic

rsaezr, 2019-08-22:
Buenos días,

Estoy teniendo un problema en la instalación de Aerostack, y es que en el último paso tras ejecutrar el script para el proyecto /teleoperation-drone-simulator: _~/temp/install_project.sh projects/teleoperation_drone_simulator_
Me aparece el siguiente mensaje cuando lleva un 87% de instalación:
Scanning dependencies of target droneTrajectoryControllerROSModule
_[ 87%] Building CXX object aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/src/source/droneTrajectoryControllerROSModule.cpp.o
In file included from /home/roberto/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/src/source/droneTrajectoryControllerROSModule.cpp:1:0:
**/home/roberto/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/src/include/droneTrajectoryControllerROSModule.h:35:10: fatal error: droneTrajectoryControllerROSModule/trajectoryControllerConfig.h: No existe el archivo o el directorio
 #include <droneTrajectoryControllerROSModule/trajectoryControllerConfig.h>**
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/build.make:62: recipe for target 'aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/src/source/droneTrajectoryControllerROSModule.cpp.o' failed
make[2]: *** [aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/src/source/droneTrajectoryControllerROSModule.cpp.o] Error 1
CMakeFiles/Makefile2:22783: recipe for target 'aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/all' failed
make[1]: *** [aerostack_stack/stack/motion_control_system/trajectory_controller/droneTrajectoryControllerROSModule/CMakeFiles/droneTrajectoryControllerROSModule.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed_

He marcado en negrita el componente que no encuentra que es la dependencia #include <droneTrajectoryControllerROSModule/trajectoryControllerConfig.h> incluida en el fichero droneTrajectoryControllerROSModule.h. He comprobado la ruta, y efectivamente no existe ningún trajectoryControllerConfig.h.

No sé si el error se debe a que este fichero no está incluido en el repositorio o es que la descarga que he hecho no ha sido correcta, pero en la descarga no ha habido ningún error. ¿Sabéis a qué puede ser debido el problema?

Muchas gracias,
Roberto Sáez.

seanfurman, 2019-10-04:
I am having the same issue installing Aerostack. I am trying to install Aerostack 3.0 on Ubuntu 18.04 with ROS Melodic. I followed the build instructions on the wiki and I got the same error after running the install_full_aerostack.sh script. 

Any help would be greatly appreciated. 

Thank you,
Sean

rodelgo7al, 2019-10-10:
Hello, try these comands:

`cd $AEROSTACK_WORKSPACE`
`catkin_make droneTrajectoryControllerROSModule_gencfg`

Then, compile the whole project again:

`cd $AEROSTACK_WORKSPACE`
`catkin_make`

seanfurman, 2019-10-28:
That worked! Thank you!

closed 2019-10-31 by rodelgo7al