cvar-upm/aerostack #45: roscore/master is not yet running, will wait for it to start

ibrahimqazi, 2019-09-04:
Hello sir!
When I run one of the example projects like:
**teleoperation_drone_simulator** In one terminal I already launched **roscore**  and in another terminal I launched `./main_launcher.sh` but after launching this file I get the following error in all terminal that **roscore is not running,** while its already running. What could be the possible reason. Please help.

rodelgo7al, 2019-09-05:
Hello, 
We need more information about your system, it can help us to find a solution. It might be a problem related to ROS configuration. Can you send us what the roscore terminal is printing? Also, by default, the ROS_MASTER_URI key environment variable is "http://localhost:11311". 

Type `echo $ROS_MASTER_URI` in a terminal and check this variable.

ibrahimqazi, 2019-09-05:
@rodelgo7al yes, it is 

> http://localhost:11311
And the contents of roscore terminal are:
... logging to /home/ibrahim/.ros/log/ae30591a-cfe3-11e9-85f0-001e101f0000/roslaunch-ibrahim-aspire-v5-591g-3940.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:38859/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [3952]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to ae30591a-cfe3-11e9-85f0-001e101f0000
process[rosout-1]: started with pid [3972]
started core service [/rosout]

Arsalan66, 2019-09-11:
have you found the answer, if so please suggest me its solution as well

ibrahimqazi, 2019-09-11:
waiting for the response of @rodelgo7al

Arsalan66, 2019-09-12:
Salam brother, have you found solution for libeigen3-dev ? if you have kindly share as i am encountering the same problem

ibrahimqazi, 2019-09-12:
> Salam brother, have you found solution for libeigen3-dev ? if you have kindly share as i am encountering the same problem

sorry brother, i searched on google but I forgot, can you share your error, may be i can remember and help you.

rodelgo7al, 2019-09-12:
@ibrahimqazi I've seen you are using ROS kinetic, if you are using Aerostack and ubuntu 18 you must install and use ros melodic. Please, update ROS and let me know if it worked.

Arsalan66, 2019-09-12:
What about ubuntu 16.04 with Ros kinetic

Arsalan66, 2019-09-12:
@rodelgo7al

ibrahimqazi, 2019-09-18:
> 
> 
> It's working on ros melodic, brother

But I don't have melodic version :(

Arsalan66, 2019-09-18:
Search how to dual boot the system on YouTube, take around 1 hour or more than that depending on the system

Arsalan66, 2019-09-18:
I have my data retained and I'm also using 18.04 with 16.04 via dual boot

ibrahimqazi, 2019-09-18:
> 
> 
> I have my data retained and I'm also using 18.04 with 16.04 via dual boot

you mean to install multiple operating systems ?

Arsalan66, 2019-09-18:
Yes multiple OS on same computer

Arsalan66, 2019-09-18:
Search how to download ubuntu 18.04 you'll have a 1.9 gb file then write how to dual boot ubuntu

QiuMingZS, 2020-02-04:
I have met this problem several days ago, while before this Aerostack runs well.
Today, I have done the follow steps, and the problem was solved.
1.Set my IP to manual, accoding to currently IP, using command 'ifconfig' in terminal to check
2.Set my ROS IP as the IP mentiond before 
'export ROS_IP=192.168.1.110', this line is in file  .bashrc
3.the problem was solved

ibrahimqazi, 2020-02-04:
@QiuMingZS can you please specify the exact commands ? because 'export ROS_IP=192.168.1.110' is not working. It says `bash: export: `ROS_IP = 192.168.1.23': not a valid identifier`

QiuMingZS, 2020-02-05:
> @QiuMingZS can you please specify the exact commands ? because 'export ROS_IP=192.168.1.110' is not working. It says `bash: export: `ROS_IP = 192.168.1.23': not a valid identifier`

using this command ->   sudo vim .bashrc
and you can see 'export ROS_IP=192.168.1.110' is a line in the file(.bashrc)

closed 2020-11-28 by martin-molina