cvar-upm/aerostack #46: Problem with gnome terminal

Arsalan66, 2019-09-11:
I've downloaded another terminal and updated the existing one but still getting the following error
"Option "--title" is no longer supported in this version of gnome-terminal.
ERROR: Unable to communicate with master!"
Please advise me the necessary steps to solve the problem.

Regards

Arsalan66, 2019-09-11:
./main_launcher.sh
-Setting ROS workspace
-Setting roscore in localhost
-Setting droneId = 111
-Swarm Drone ID = 1
-Setting droneIp = 192.168.1.1
- Waiting for all process to be started...
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
Option "--title" is no longer supported in this version of gnome-terminal.
ERROR: Unable to communicate with master!
ros@ros-HP-Pavilion-dv6-Notebook-PC:~/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/projects/teleoperation_drone_simulator$ Option "--title" is no longer supported in this version of gnome-terminal.

rodelgo7al, 2019-09-12:
Hello @Arsalan66  , which version of ubuntu are you using?
Also, I would like to know which version of gnome-terminal is running.
Type: 
`gnome-terminal --version`

You can check/change your current terminal typing:
`sudo update-alternatives --config x-terminal-emulator`

By the way, as a temporal solution, you can modify the main_launcher.sh file and remove each '--title "<string>" ' 

Finally, to solve the error "Unable to communicate with master!" you must have installed ROS melodic (Ubuntu 18) and run "roscore" before executing the main_launcher.sh file.

closed 2019-09-18 by Arsalan66