cvar-upm/aerostack #48: problem with rosservice

Arsalan66, 2019-10-01:
Salam to Hi , I'm encountering a problem in basic mission with drone simulator when i write the command:  rosservice call /drone111/python_based_mission_interpreter_process/start
It says:ERROR: Service [/drone111/python_based_mission_interpreter_process/start] is not available.
Although project modules were installed correctly 
Any help/assistance will be highly appreciated

rodelgo7al, 2019-10-01:
Hello, make sure you have all files related to python_based_mission_interpreter in this path: 
`workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process`

You should have there some files and folders (launch, src...) to execute that process.

Also, check that there isn't any **CATKIN_IGNORE** file in that folder or any folder above that one.

Compile:
`cd $AEROSTACK_WORKSPACE` 
`catkin_make`

Try to execute the project again. If the error persist, check aerostack terminals after running "main_launcher.sh" to see if any of them has an error.

Arsalan66, 2019-10-01:
Thank you so much brother, really means a lot

closed 2019-10-02 by rodelgo7al