cvar-upm/aerostack #50: Problem in saving custom behavior tree

Arsalan66, 2019-10-02:
[image]

Salam and hi , I'm encountering a problem in creating a custom behaviour tree in project Behavior tree mission with "Drone simulator", when I launch this command $ ./mission_editor_launcher.sh
It outputs results as shown in picture and does not save in yaml file
Any help/assistance to solve problem would be highly appreciated

rodelgo7al, 2019-10-02:
Hello, export your tree ("Export" button) with the name "behavior_tree_mission_file" (without .yaml) to your documents folder or desktop. Delete the old behavior_tree_mission_file.yaml located in /mission/ and replaced it with the new one.

Arsalan66, 2019-10-02:
Thanks for response, I'll let you know the progress

Arsalan66, 2019-10-02:
Thanks brother , it worked

closed 2019-10-02 by Arsalan66