cvar-upm/aerostack #51: rosservice error fix

Arsalan66, 2019-10-02:
After checking that there is no CATKIN_IGNORE file in :
workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process
I ran cd $AEROSTACK_WORKSPACE
catkin_make
It compiled successfully however there was an error in ther terminals :+1: 

"
**process[drone111/python_based_mission_interpreter_process-1]: started with pid [10815]
Traceback (most recent call last):
  File "/home/arsalan/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process/src/pml_mission_interpreter_process.py", line 7, in <module>
    import mission_handler
  File "/home/arsalan/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process/src/mission_handler.py", line 2, in <module>
    import mission_loader
  File "/home/arsalan/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process/src/mission_loader.py", line 2, in <module>
    import pyinotify
ImportError: No module named pyinotify
[drone111/python_based_mission_interpreter_process-1] process has died [pid 10815, exit code 1, cmd /home/arsalan/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process/src/pml_mission_interpreter_process.py __name:=python_based_mission_interpreter_process __log:=/home/arsalan/.ros/log/042d38d4-e514-11e9-81d7-701a040a8ed4/drone111-python_based_mission_interpreter_process-1.log].
log file: /home/arsalan/.ros/log/042d38d4-e514-11e9-81d7-701a040a8ed4/drone111-python_based_mission_interpreter_process-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete**
"

[image]
****
I'd really appreciate some assistance to remove this issue

Arsalan66, 2019-10-02:
[image]

rodelgo7al, 2019-10-03:
You must install pyinotify. Try these options:

```
sudo apt-get install python-pyinotify
sudo dpkg-reconfigure python-pyinotify
```
If it doesn't work:
```
git clone https://github.com/seb-m/pyinotify.git
cd pyinotify/
sudo python3.6 setup.py install 
```

Arsalan66, 2019-10-03:
Thanks for the generous reply ,I'll inform you about the progress

closed 2019-10-08 by Arsalan66