cvar-upm/aerostack #57: issues running basic_lidar_rotors_simulator project

furmans, 2019-10-28:
Hello, 

I have been having some issues getting the project basic_lidar_rotors_simulator to run successfully. When running the project, the drone does not complete all of the actions that are in the example YouTube video of the project. In addition, I have received multiple error messages and warnings on the terminals that are brought up when the project is launched. The picture below is a screen shot of the execution viewer while the project was running. This was the final state of the execution viewer during the test and the drone does not move much past the initial takeoff. The second figure shows the final location of the drone in the Gazebo simulator

[
[image]
](url)

[
[image]
](url)

I am running Ubuntu 16.04.6 LTS, ROS Kinetic, and the latest Aerostack release. Also, I am running everything within a virtual machine on a Linux host. I have allocated 14 GB RAM and 3 CPUs to the virtual machine. 

In order to run the project, I re-added the following submodules: resource_manager, and resource_manager_process

When I first ran the project, there were a few error messages that appeared that I was able to resolve on my own. 

[ERROR] [1572010582.620071307]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..

[ERROR] [1572010569.166114432, 11.810000000]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_worldmodel_geotiff_plugins/hector_geotiff_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..

I fixed the above two errors by cloning the hector_localization and hector_slam repositories and copying the referenced XML files into their expected locations. 

I also had the following error message 

Usage: roslaunch [options] [package] <filename> [arg_name:=value...]
       roslaunch [options] <filename> [<filename>...] [arg_name:=value...]

If <filename> is a single dash ('-'), launch XML is read from standard input.

roslaunch: error: The following input files do not exist:  
bash: line 2: drone_id:=111: command not found

I was able to fix this by changing the project’s main_launcher.sh script. I believe the script had a small syntax error in the section where it launches the ‘self localization process selector’ and I just copied that section from the main_launcher.sh script of another project and the error went away. 

Currently, I am seeing the following warnings/errors when I run the project: 

From one of the Gazebo terminals: 

Master Unknown message type[] From[55260]

[libprotobuf ERROR google/protobuf/message_lite.cc:123] Can't parse message of type "gazebo.msgs.Packet" because it is missing required fields: stamp, type, serialized_data

 On another terminal: 

[WARNING]: The following behaviors are set in the catalog but not launched: behavior_follow_path_with_pid_control, behavior_generate_path_with_occupancy_grid, behavior_go_to_point_with_occupancy_grid, behavior_keep_hovering_with_pid_control, behavior_keep_moving_with_pid_control, behavior_land, behavior_localize_with_odometry, behavior_rotate_with_pid_control, behavior_self_localize_and_map_with_lidar, behavior_take_off, behavior_wait

From tab 6 of the terminal with about 15 tabs: 

[ WARN] [1572288785.800090005, 21.200000000]: Map update loop missed its desired rate of 50.0000Hz... the loop actually took 0.0300 seconds

[ WARN] [1572288812.463420002, 31.660000000]: MessageFilter [target=map hokuyo::sensors_link ]: Dropped 100.00% of messages so far. Please turn the [ros.costmap_2d.message_notifier] rosconsole logger to DEBUG for more information.

[ WARN] [1572288831.201980773, 38.410000000]: Costmap2DROS transform timeout. Current time: 38.4100, global_pose stamp: 38.1000, tolerance: 0.3000

[ WARN] [1572288831.203056734, 38.410000000]: Could not get robot pose, cancelling reconfiguration

*These messages are repeated multiple times

When I updated the log level, I received the following log messages

[DEBUG] [1572027401.929180348, 95.570000000]: MessageFilter [target=map hokuyo::sensors_link ]: Added message in frame hummingbird_laser1/hokuyo_link at time 95.510, count now 50

[DEBUG] [1572027401.929321299, 95.570000000]: MessageFilter [target=map hokuyo::sensors_link ]: Removed oldest message because buffer is full, count now 50 (frame_id=hummingbird_laser1/hokuyo_link, stamp=93.970000)   

From tab 7 of the terminal with about 15 tabs: 

Warning: TF_OLD_DATA ignoring data from the past for frame hummingbird_laser1/hokuyo_link at time 0.1 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp

[ERROR] [1572288767.555207308, 15.870000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation into the past.  Requested time 15.080000000 but the earliest data is at time 15.810000000, when looking up transform from frame [base_link] to frame [odom]

[ERROR] [1572288768.391085649, 16.370000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation at time 15.690000000, but only time 15.080000000 is in the buffer, when looking up transform from frame [base_link] to frame [odom]

Any help would be greatly appreciated. 

Thanks in advance, 
Sean

furmans, 2019-10-29:
Output of rosrun tf view_frames
[frames.pdf](https://github.com/Vision4UAV/Aerostack/files/3785006/frames.pdf)

Output of command

furmans, 2019-10-29:
Output from rosrun tf view_frames: 

[frames.pdf](https://github.com/Vision4UAV/Aerostack/files/3785290/frames.pdf)

Output from roswtf: 

Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 3 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /drone111/process_monitor_process_main:
   * /drone111/self_detected_process_error
 * /drone111/droneCommand:
   * /drone111/battery
 * /drone111/hector_mapping:
   * /drone111/syscommand
 * /drone111/droneRobotLocalizationROSModuleNode:
   * /drone111/mavros/global_position/raw/fix
   * /drone111/lb_estimated_pose/door_pose
   * /drone111/states/ardrone3/PilotingState/AltitudeChanged
   * /final_pose
   * /drone111/odom
   * /drone111/vb_estimated_pose/rpnp_pose
   * /pose
   * /drone111/lb_estimated_pose/window_pose
   * /drone111/states/ardrone3/PilotingState/SpeedChanged
   * /drone111/states/ardrone3/PilotingState/AttitudeChanged
 * /drone111/belief_updater_process:
   * /drone111/qr_interpretation
   * /drone111/battery
   * /drone111/arucoObservation
 * /drone111/hector_trajectory_server:
   * /drone111/syscommand
 * /drone111/imu_attitude_to_tf_node:
   * /drone111/imu_topic
 * /drone111/hector_geotiff_node:
   * /drone111/syscommand
 * /drone111/droneInterface:
   * /drone111/get_bounding_box
   * /drone111/tracking_object
   * /drone111/camera/front/image_raw
   * /drone111/SOEstimatedSpeeds
   * /drone111/ArucoSlam_EstimatedSpeeds
   * /drone111/controller_image_features
   * /drone111/magnetometer
   * /drone111/controller_image_features_feedback
   * /drone111/SOEstimatedPose
   * /drone111/is_object_on_frame
   * /drone111/battery
   * /drone111/ArucoSlam_EstimatedPose
 * /drone111/move_base:
   * /drone111/odom
   * /drone111/move_base/cancel
 * /rviz_1572370891110703815:
   * /drone111/exploration_polygon_marker
   * /drone111/map_updates
   * /drone111/exploration_polygon_marker_array
 * /drone111/droneTrajectoryController:
   * /drone111/command_PBVS/Altitude
   * /drone111/EstimatedPose_iRobotGMR_wrt_droneGMR
 * /gazebo:
   * /hummingbird_laser1/wind_speed
   * /gazebo/set_model_state
   * /gazebo/set_link_state

WARNING The following nodes are unexpectedly connected:
 * /drone111/imu_attitude_to_tf_node->/roswtf_4492_1572370914222 (/tf)
 * /drone111/amcl->/roswtf_4492_1572370914222 (/tf)
 * /drone111/amcl->/roswtf_6854_1572370936133 (/tf)
 * /drone111/imu_attitude_to_tf_node->/roswtf_6854_1572370936133 (/tf)
 * /drone111/droneRobotLocalizationROSModuleNode->/roswtf_9435_1572370965215 (/tf)
 * /drone111/amcl->/roswtf_9435_1572370965215 (/tf)
 * /drone111/imu_attitude_to_tf_node->/roswtf_9435_1572370965215 (/tf)
 * /drone111/droneRobotLocalizationROSModuleNode->/roswtf_4492_1572370914222 (/tf)
 * /drone111/droneRobotLocalizationROSModuleNode->/roswtf_6854_1572370936133 (/tf)

WARNING These nodes have died:
 * hummingbird_laser1/spawn_hummingbird_laser1-3

Found 1 error(s).

ERROR The following nodes should be connected but aren't:
 * /drone111/move_base->/drone111/move_base (/drone111/move_base/local_costmap/footprint)
 * /drone111/move_base->/drone111/move_base (/drone111/move_base/global_costmap/footprint)

furmans, 2019-11-04:
Update: 

I had originally missed the following error when running the project: 

AttributeError: 'module' object has no attribute 'consultBelief'
[ERROR] Runtime error encountered in mission
[drone111/python_based_mission_interpreter_process-1] killing on exit

I resolved this error by checking out to the previous commit on the python_based_mission_interpretor submodule. 

Now when I run the project, the drone attempts to navigate to the first destination point, but crashes prior to reaching it. (it generates a path to the first point and attempts to follow path with PID control). The drone's movements are usually much quicker (less stable) than in the demonstration video.  

Here are the warnings/error mesages that I currently get when I run the project. 

Terminal running 'main_launcher.sh' 

 MessageFilter [target=map ]: Discarding message from [/drone111/move_base] due to empty frame_id.  This message will only print once.
Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty

Another terminal
[WARNING]: The following behaviors are set in the catalog but not launched: behavior_follow_path_with_pid_control, behavior_generate_path_with_occupancy_grid, behavior_go_to_point_with_occupancy_grid, behavior_keep_hovering_with_pid_control, behavior_keep_moving_with_pid_control, behavior_land, behavior_localize_with_odometry, behavior_rotate_with_pid_control, behavior_self_localize_and_map_with_lidar, behavior_take_off, behavior_wait

Terminal in Tab 3
Warning: TF_OLD_DATA ignoring data from the past for frame hummingbird_laser1/hokuyo_link at time 0.07 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp

Terminal in Tab 5
referenceFrames::getxyzYPRfromHomogMatrix_wYvPuR: ¡Bad conditioning in Pitch! Pitch =+-pi/2

Terminal in Tab 6
[ WARN] [1572884140.788252444, 32.650000000]: Map update loop missed its desired rate of 50.0000Hz... the loop actually took 0.0400 seconds
[ WARN] [1572884142.379002342, 33.260000000]: MessageFilter [target=map hokuyo::sensors_link ]: Dropped 100.00% of messages so far. Please turn the [ros.costmap_2d.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1572884156.257767393, 39.350000000]: Costmap2DROS transform timeout. Current time: 39.3500, global_pose stamp: 39.0400, tolerance: 0.3000
[ WARN] [1572884158.122325612, 40.170000000]: Unable to get starting pose of robot, unable to create global plan
[ WARN] [1572884158.183900222, 40.210000000]: Could not get robot pose, cancelling reconfiguration

*Each of these errors is printed multiple times 

Terminal in Tab 7
[ERROR] [1572884101.580454377, 17.680000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation into the past.  Requested time 16.830000000 but the earliest data is at time 16.990000000, when looking up transform from frame [base_link] to frame [odom]

*This error was printed just once or twice

Output of Tab 14
process[drone111/python_based_mission_interpreter_process-1]: started with pid [29178]
[INFO] Mission loaded
Starting mission...
Taking off...
TAKE_OFF
Localizating off...
[ERROR] 
Moving...
GENERATE_PATH_WITH_OCCUPANCY_GRID
............
path (1,?y)
True
137
[[-0.0249977, -0.0249977, 1], [-0.0574379, -0.02855, 1], [-0.0820412, -0.0329933, 1], [-0.102207, -0.0182167, 1], [-0.122535, -0.00366592, 1], [-0.140839, 0.0133629, 1], [-0.151774, 0.0358482, 1], [-0.165369, 0.056826, 1], [-0.182413, 0.0751122, 1], [-0.173969, 0.0986473, 1], [-0.166709, 0.122567, 1], [-0.153764, 0.143954, 1], [-0.142844, 0.166445, 1], [-0.138728, 0.191103, 1], [-0.134187, 0.215688, 1], [-0.130738, 0.24045, 1], [-0.125977, 0.264993, 1], [-0.119285, 0.289077, 1], [-0.110651, 0.312539, 1], [-0.0989906, 0.334652, 1], [-0.0861488, 0.356106, 1], [-0.0727303, 0.3772, 1], [-0.0584724, 0.397732, 1], [-0.0434822, 0.417739, 1], [-0.028434, 0.437704, 1], [-0.0131203, 0.457467, 1], [0.00226669, 0.477169, 1], [0.0181847, 0.496444, 1], [0.0341393, 0.515694, 1], [0.0502403, 0.534817, 1], [0.0667809, 0.553567, 1], [0.0836449, 0.572024, 1], [0.100491, 0.590493, 1], [0.117617, 0.608706, 1], [0.134866, 0.626803, 1], [0.152254, 0.644765, 1], [0.169753, 0.662618, 1], [0.187258, 0.680471, 1], [0.20483, 0.698251, 1], [0.222463, 0.715969, 1], [0.240054, 0.733736, 1], [0.257705, 0.751437, 1], [0.275497, 0.769003, 1], [0.293154, 0.786697, 1], [0.310873, 0.804342, 1], [0.32875, 0.82181, 1], [0.346505, 0.839413, 1], [0.364297, 0.856979, 1], [0.382229, 0.874398, 1], [0.400045, 0.891933, 1], [0.417892, 0.909438, 1], [0.435732, 0.926949, 1], [0.453597, 0.944442, 1], [0.471481, 0.96191, 1], [0.489364, 0.979385, 1], [0.507259, 0.996841, 1], [0.525277, 1.01417, 1], [0.543191, 1.03161, 1], [0.561117, 1.04904, 1], [0.579134, 1.06636, 1], [0.597073, 1.08377, 1], [0.615023, 1.10118, 1], [0.633047, 1.1185, 1], [0.651003, 1.1359, 1], [0.668972, 1.15328, 1], [0.687002, 1.1706, 1], [0.704977, 1.18797, 1], [0.722964, 1.20534, 1], [0.741, 1.22265, 1], [0.758993, 1.24, 1], [0.777065, 1.25728, 1], [0.795107, 1.27458, 1], [0.813119, 1.29192, 1], [0.831191, 1.30919, 1], [0.849215, 1.32652, 1], [0.867239, 1.34384, 1], [0.885317, 1.36111, 1], [0.903353, 1.37843, 1], [0.921389, 1.39574, 1], [0.939474, 1.413, 1], [0.957522, 1.43029, 1], [0.975588, 1.44757, 1], [0.993655, 1.46486, 1], [1.01171, 1.48215, 1], [1.02918, 1.50003, 1], [1.04693, 1 [... cut ...]
[ERROR] [1572880911.088798124, 68.670000000]: Service call failed: service [/drone111/quadrotor_motion_with_trajectory_controller/behavior_follow_path_with_pid_control/activate_behavior] responded with an error: yaml-cpp: error at line 0, column 0: bad conversion

The following error has also been logged by the quadrotor_motion_with_trajectory_controller  process. 

[ERROR] [1572880911.088626516, 68.670000000]: Exception thrown while processing service call: yaml-cpp: error at line 0, column 0: bad conversion

raulcrbe, 2019-12-05:
I remember that this error "error: yaml-cpp: error at line 0, column 0: bad conversion" was something related with the arguments of the behavior, I do not know if it is the name of the argument "path" which can differ from the one that the behavior wants, or if it is the content of the argument, maybe it wants another type of array or something.

I hope this can help you.

closed 2020-11-28 by martin-molina