cvar-upm/aerostack #58: Installation error

infiBoy, 2019-11-02:
CMake Error at aerostack_stack/stack/hardware_interface/drivers_platforms/driver_pixhawk/mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

furmans, 2019-11-04:
I'm not sure if this will fix your issue, but see if the ROS package mavlink is installed on your system. 
You can check by running 'dpkg -l | grep mavlink'. If it doesn't show up you can run 'apt-cache search mavlink' and then install the mavlink package. Then try building again.

pasanci, 2019-11-12:
Hello,
As @furmans sugests, your error is fixed installing the ROS package mavlink, just run 'apt-get install ros-$ROS-DISTRO-mavlink'  and recompile

closed 2021-02-17 by pasanci