cvar-upm/aerostack #60: ERROR when executing project "warehouse_rotors_simulator"

QiuMingZS, 2020-02-05:
I have installed AEROSTACK 3.0 successfully and the project "bridge_inpspection_rotors_simulator"  works well.
For project "warehouse_rotors_simulator", Gazebo is ok. But when start "./main_launcher.sh",there are some errors, listed below:

1.
[attention_to_visual_markers.launch] is neither a launch file in package [attention_to_visual_markers] nor is [attention_to_visual_markers] a launch file name
The traceback for the exception was written to the log file
2.
[navigation_with_lidar.launch] is neither a launch file in package [navigation_with_lidar] nor is [navigation_with_lidar] a launch file name
The traceback for the exception was written to the log file
3.
[qr_recognizer.launch] is neither a launch file in package [qr_recognizer] nor is [qr_recognizer] a launch file name
The traceback for the exception was written to the log file
4.
[path_planner_in_occupancy_grid.launch] is neither a launch file in package [path_planner_in_occupancy_grid] nor is [path_planner_in_occupancy_grid] a launch file name
The traceback for the exception was written to the log file
5.
[drone111/python_based_mission_interpreter_process-1] process has died [pid 19020, exit code 1, cmd /home/fire/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/mission_control_system/python_based_mission_interpreter/python_based_mission_interpreter_process/src/pml_mission_interpreter_process.py __name:=python_based_mission_interpreter_process __log:=/home/fire/.ros/log/ed066a5c-47b2-11ea-99ce-e094677626e5/drone111-python_based_mission_interpreter_process-1.log].
log file: /home/fire/.ros/log/ed066a5c-47b2-11ea-99ce-e094677626e5/drone111-python_based_mission_interpreter_process-1*.log

I have executed "source devel/setup.bash", but errors still exist.

rodelgo7al, 2020-02-05:
Did you install a full version of Aerostack with the command

~/temp/install_full_aerostack.sh

Or did you install a specific project?

It seems that you don't have installed a full Aerostack version. Check if your workspace has this file: "navigation_with_lidar.launch" and if you find it, delete the CATKIN_IGNORE file in that folder. If you can't find that file, I'd recommend you to delete your workspace and run the command above.

Also, "python_based_mission_interpreter_process" may have died because "pyinotify" isn't installed or the project doesn't find the mission.py file. It would be helpful if you can provide more information about that error (it usually gives more information before reaching "process has died" error).

QiuMingZS, 2020-02-05:
I installed the full version using this command and just now deleted three CATKIN_IGNORE files and solved some errors.
Still have some errors below:
1.
[ERROR] [1580896216.756078028]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..

fully information is :
WARNING: Package name "droneVisualMarkersLocalizer" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "driverRotorsSimulatorROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneEKFStateEstimator" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneEKFStateEstimatorROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneSpeedPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "dronePBVSPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneRobotLocalizationROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneModuleInterfaceROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryStateMachine" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMidLevelAutopilot" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneInterfaceROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneModuleROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMsgsROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryControllerROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "driver_px4flow_interface_ROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "lib_cvgloggerROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneLoggerROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "referenceFramesROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "referenceFrames" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMidLevelAutopilotROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
master has started, initiating launch
... logging to /home/fire/.ros/log/d633ed32-47fc-11ea-99ce-e094677626e5/roslaunch-wargod-23874.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.110:39339/

SUMMARY
========

PARAMETERS
 * /drone111/attention_to_visual_markers/attention_to_visual_markers_manager/activate_behavior_srv: activate_behavior
 * /drone111/attention_to_visual_markers/attention_to_visual_markers_manager/behavior_system: attention_to_visu...
 * /drone111/attention_to_visual_markers/attention_to_visual_markers_manager/check_activation_conditions_srv: check_activation_...
 * /drone111/attention_to_visual_markers/attention_to_visual_markers_manager/consult_belief: consult_belief
[... cut ...] ected trees.
[ERROR] [1580896212.911406544, 1631.107000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation at time 1630.469000000, but only time 1630.068000000 is in the buffer, when looking up transform from frame [base_link] to frame [odom]
[ INFO] [1580896212.968808700, 1631.142000000]: Finished waiting for tf, waited 1.685000 seconds
path_planner_in_occupancy_grid.launch

Besides, what is CATKIN_IGNORE file used for and why it exsits?
thank you.

rodelgo7al, 2020-02-05:
Try executing these lines (I suppose you have installed ubuntu 16 and ros Kinetic):

```
sudo apt-get install ros-kinetic-move-base*

sudo apt-get install ros-kinetic-hector*

sudo apt-get install ros-kinetic-amcl*
```
And run "./lidar_instalation.sh" script.

Answering your question, CATKIN_IGNORE avoids compiling the folder this file is included in, so basically, it makes compilation faster when we exclude some folders that we are not going to use. Therefore, Aerostack can't find files if they are not compiled.

QiuMingZS, 2020-02-05:
> Try executing these lines (I suppose you have installed ubuntu 16 and ros Kinetic):
>
> ```
> sudo apt-get install ros-kinetic-move-base*
>
> sudo apt-get install ros-kinetic-hector*
>
> sudo apt-get install [...]

Thank you and I installed ubuntu 16 and ros kinetic.
I did these commands, but still have an error:

[ERROR] [1580903599.255036057, 1628.321000000]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_worldmodel_geotiff_plugins/hector_geotiff_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..

fully information:
```
master has started, initiating launch
... logging to /home/fire/.ros/log/0fdd89d8-480e-11ea-b03a-e094677626e5/roslaunch-wargod-2109.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "droneVisualMarkersLocalizer" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "driverRotorsSimulatorROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneEKFStateEstimator" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneEKFStateEstimatorROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneSpeedPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "dronePBVSPositionMidLevelController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneRobotLocalizationROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneModuleInterfaceROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryController" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryStateMachine" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMidLevelAutopilot" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneInterfaceROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneModuleROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMsgsROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneTrajectoryControllerROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "driver_px4flow_interface_ROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "lib_cvgloggerROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneLoggerROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "referenceFramesROS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "referenceFrames" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "droneMidLevelAutopilotROSModule" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://192.168.1.110:43824/

SUMMARY
========

PARAMETERS
 * /drone111/hector_geotiff_node/draw_background_checkerboard: True
 * /drone111/hector_geotiff_node/draw_free_space_grid: True
 * /drone111/hector_geotiff_node/geotiff_save_period: 0.0
 * /drone111/hector_geotiff_node/map_file_base_name: hector_slam_map
 * /drone111/hector_geotiff_node/map_file_path: /opt/ros/kinetic/...
[... cut ...]
[ERROR] [1580903600.436748143, 1629.284000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation into the past.  Requested time 1628.752000000 but the earliest data is at time 1629.161000000, when looking up transform from frame [base_link] to frame [odom]
[ INFO] [1580903601.046945088, 1629.766000000]: Finished waiting for tf, waited 1.612000 seconds
[ERROR] [1580903601.100496187, 1629.788000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation at time 1629.127000000, but only time 1628.752000000 is in the buffer, when looking up transform from frame [base_link] to frame [odom]

```

QiuMingZS, 2020-02-05:
I have solved this error by add the missing file, 
```
[ERROR] [1580903599.255036057, 1628.321000000]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_worldmodel_geotiff_plugins/hector_geotiff_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
```
but still exists 2 errors;
```
[ERROR] [1580903600.436748143, 1629.284000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation into the past.  Requested time 1628.752000000 but the earliest data is at time 1629.161000000, when looking up transform from frame [base_link] to frame [odom]
[ INFO] [1580903601.046945088, 1629.766000000]: Finished waiting for tf, waited 1.612000 seconds
[ERROR] [1580903601.100496187, 1629.788000000]: Transform failed during publishing of map_odom transform: Lookup would require extrapolation at time 1629.127000000, but only time 1628.752000000 is in the buffer, when looking up transform from frame [base_link] to frame [odom]

```

closed 2020-11-28 by martin-molina