cvar-upm/aerostack #63: how to solve Cmake error

manivara507, 2020-04-10:
[image]

rodelgo7al, 2020-04-13:
Try to install some ros dependencies:

`sudo apt-get install ros-melodic-octomap ros-melodic-octomap-mapping ros-melodic-octomap-msgs ros-melodic-octomap-ros ros-melodic-octomap-rviz-plugins ros-melodic-octomap-server`

ted-15, 2020-04-13:
[image]

I am facing similar error also i have attached the screenshot of the same. Due to this error configuration is becoming incomplete and shows cmake error. Help me with finding resolution of it.

Thanks in advance.

rodelgo7al, 2020-04-14:
I think you should reinstall ros melodic and Aerostack. Delete workspace and temp folder and execute this command:

> sudo apt-get remove ros-*

Then follow these steps: http://wiki.ros.org/melodic/Installation/Ubuntu

And reinstall Aerostack. Thank you.

ted-15, 2020-04-19:
[image]

i removed ros and installed it again. also i followed the installation procedure given but still i get an error and it says configuration incomplete. I have attached the screenshot 

how to resolve it as i am tried many times still i get the same error.

Thanks

ted-15, 2020-04-19:
i have fixed previous error using the command sudo apt-get install ros-$ROS_DISTRO-mavlink and then followed by an another error 

[image]

and then i used this command to fix it sudo apt-get install ros-melodic-octomap ros-melodic-octomap-mapping ros-melodic-octomap-msgs ros-melodic-octomap-ros ros-melodic-octomap-rviz-plugins ros-melodic-octomap-server

after which i am facing the current error about Could not find a package configuration file provided by "opencv_apps" with
  any of the following names:

    opencv_appsConfig.cmake
    opencv_apps-config.cmake

i have attached screenshot also.
[image]

help me to resolve it.

ted-15, 2020-04-20:
[image]

Also i have fixed the previous errors. now i am facing an new error for which i have attached the screenshot. also the error below 

[ 74%] Generating qrc_images.cpp
RCC: Error in '/home/solomon/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/src/view/resources/images.qrc': Cannot find file 'images/Kompaß360.png'
aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/build.make:134: recipe for target 'aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/qrc_images.cpp' failed
make[2]: *** [aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/qrc_images.cpp] Error 1
CMakeFiles/Makefile2:44179: recipe for target 'aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/all' failed
make[1]: *** [aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 74%] Linking CXX shared library /home/solomon/workspace/ros/aerostack_catkin_ws/devel/lib/liblib_cvgloggerROS.so
[ 74%] Built target lib_cvgloggerROS
[ 74%] Building CXX object aerostack_stack/stack/execution_control_system/behavior_systems/navigation_with_lidar/CMakeFiles/navigation_with_lidar.dir/src/source/behavior_generate_path_with_occupancy_grid.cpp.o
[ 74%] Linking CXX shared library /home/solomon/workspace/ros/aerostack_catkin_ws/devel/lib/libnavigation_with_lidar.so
[ 74%] Built target navigation_with_lidar
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

kindly help me to resolve me this error. Thanks in advance.

rodelgo7al, 2020-04-23:
Which ubuntu version are you using?

ted-15, 2020-04-23:
I am using Ubuntu 18.04 lts

ted-15, 2020-04-28:
Also i have fixed the previous errors. now i am facing an new error for which i have attached the screenshot. also the error below

[ 74%] Generating qrc_images.cpp
RCC: Error in '/home/solomon/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/src/view/resources/images.qrc': Cannot find file 'images/Kompaß360.png'
aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/build.make:134: recipe for target 
'aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/qrc_images.cpp' failed
make[2]: *** [aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/qrc_images.cpp] Error 1
CMakeFiles/Makefile2:44179: recipe for target 'aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/all' failed
make[1]: *** [aerostack_stack/stack/ground_control_system/graphical_user_interface/parameters_viewer_process/CMakeFiles/parameters_viewer_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 74%] Linking CXX shared library /home/solomon/workspace/ros/aerostack_catkin_ws/devel/lib/liblib_cvgloggerROS.so
[ 74%] Built target lib_cvgloggerROS
[ 74%] Building CXX object aerostack_stack/stack/execution_control_system/behavior_systems/navigation_with_lidar/CMakeFiles/navigation_with_lidar.dir/src/source/behavior_generate_path_with_occupancy_grid.cpp.o
[ 74%] Linking CXX shared library /home/solomon/workspace/ros/aerostack_catkin_ws/devel/lib/libnavigation_with_lidar.so
[ 74%] Built target navigation_with_lidar
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

please someone help me to resolve me this error. Thanks in advance.

[image]

closed 2020-11-28 by martin-molina

ted-15, 2020-11-28:
Hello Sir,

I am getting this qrc_images.cpp error while I am installing always.

Can you please help me with this to resolve? I having been doing this for
months now but still I didn't find any solution. after the new version this
error for me is not fixed.

Kindly help me out.

PFA.

Waiting for your reply.