cvar-upm/aerostack #67: issue about: Dynamic Mission Planning Architecture for UAV Swarm Coordination

MOMOHTT, 2020-06-26:
Dear professor:
  I have read your paper “A flexible and dynamic mission planning architecture for UAV swarm coordination” in details. The Dynamic Mission Planning Architecture for UAV Swarm coordination your proposed can automatically generate each UAV mission form high-level global mission. I am very interested in this research. However, after reading this paper, I still don't know how to automatically generate each UAV mission. And I can't find this module in Aerostack. So, can you help me point out this module, if this module exists in Aerostack, if not, can you provide relevant codes about generating each UAV mission form high-level global mission automatically. I think reading the source code directly will help me understand better. Thank you so much!

martin-molina, 2020-06-26:
Dear Momohtt, Thank you very much for your email and your interest in our paper. Unfortunately, the code corresponding to this work is not available in the last version of Aerostack, because the software environment has changed in the last versions. Perhaps you can email Carlos Sampedro. He may give you additional information about this. Thank you and best regards

MOMOHTT, 2020-06-26:
Thank you very much for your reply！And can your tell me what  Carlos Sampedro's email is? Thank you and best regards

| |
西北工业大学18硕-贺童
|
|
hetong8316@163.com
|
签名由网易邮箱大师定制

MOMOHTT, 2020-06-29:
Dear professor:

  Excuse me, I emailed you 3 days ago to ask about Carlos Sampedro's email。Maybe you are too busy to read my email. I have tried my best but still can’t find Carlos Sampedro's email. So if you see this email, please reply to me. Thank you and best regards!

| |
西北工业大学18硕-贺童
|
|
hetong8316@163.com
|
签名由网易邮箱大师定制

martin-molina, 2020-10-25:
Issue closed

closed 2020-10-25 by martin-molina