cvar-upm/aerostack #69: Installation error using source code

ted-15, 2020-10-02:
KIndly look into issue and give some solution. i used to face this error in earlier version as well and now with the new version still i get this error. Looking for a quick resolution. thank you. 

i have attached the error report below

[ 70%] Built target behavior_request_operator_assistance
[ 70%] Automatic MOC for target behavior_tree_editor_lib
[ 70%] Built target behavior_tree_editor_lib_autogen
[ 70%] Generating qrc_images.cpp
RCC: Error in '/home/solmon/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/user_interface/behavior_tree_editor/src/view/resources/images.qrc': Cannot find file 'images/Kompaß360.png'
aerostack_stack/stack/user_interface/behavior_tree_editor/CMakeFiles/behavior_tree_editor_lib.dir/build.make:134: recipe for target 'aerostack_stack/stack/user_interface/behavior_tree_editor/qrc_images.cpp' failed
make[2]: *** [aerostack_stack/stack/user_interface/behavior_tree_editor/qrc_images.cpp] Error 1
CMakeFiles/Makefile2:19651: recipe for target 'aerostack_stack/stack/user_interface/behavior_tree_editor/CMakeFiles/behavior_tree_editor_lib.dir/all' failed
make[1]: *** [aerostack_stack/stack/user_interface/behavior_tree_editor/CMakeFiles/behavior_tree_editor_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

pasanci, 2020-10-05:
You seem to be missing some files, try looking for the missing file "Kompaß360.png" located in the $AEROSTACK_STACK/stack/user_interface/behavior_tree_editor/src/view/resources/images, and make sure you have read permissions for all the files in this directory. Otherwise i would recommend you to delete the behavior_tree_editor folder and redownload it and recompile, execute:

  $ cd $AEROSTACK_STACK
  $ rm -r stack/user_interface/behavior_tree_editor
  $ git submodule update --init stack/user_interface/behavior_tree_editor
  $ cd $AEROSTACK_WORKSPACE
  $ catkin_make

Hope this fixes your problem.

closed 2021-02-17 by pasanci

APVGITHUB, 2021-03-08:
I got that same error, by changing the name Kompaß360.png to Kompa360.png in ~/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/user_interface/behavior_tree_editor/src/view/resources/images.qrc
and ~/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/stack/user_interface/behavior_tree_interpreter/src/view/resources/images