cvar-upm/aerostack #71: base operand of ‘->’ has non-pointer type ‘cv::Mat’

ibrahimqazi, 2020-11-23:
Can you please help me to troubleshoot this problem. I am using Ubuntu 18.04 and ROS is Melodic. When I compile the stack it gives me the following error.
```
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp: In member function ‘int CVG::Matrix::pseudoinverse(CVG::Matrix*)’:
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp:279:13: error: base operand of ‘->’ has non-pointer type ‘cv::Mat’
    sourceAux->data.fl[(fila-1)*numColumnas+columna-1]=matA->getValueData(fila,columna);
             ^~
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp:283:32: error: ‘CV_SVD’ was not declared in this scope
     cvInvert(sourceAux,pinvAux,CV_SVD); //CV_SVD
                                ^~~~~~
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp:283:32: note: suggested alternative: ‘CV_SSE’
     cvInvert(sourceAux,pinvAux,CV_SVD); //CV_SVD
                                ^~~~~~
                                CV_SSE
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp:283:5: error: ‘cvInvert’ was not declared in this scope
     cvInvert(sourceAux,pinvAux,CV_SVD); //CV_SVD
     ^~~~~~~~
/home/ibrrahim/catkin_workspaces/aerostack_catkin_ws/src/aerostack_stack/stack/libraries/lib_cvgutils/src/source/matrixLib.cpp:288:32: error: base operand of ‘->’ has non-pointer type ‘cv::Mat’
    Matrix::setValueData(pinvAux->data.fl[(fila-1)*numColumnas+columna-1],fila,columna);
```
Your help will be highly appreciated.
@jespestana @joselusl

pasanci, 2020-11-23:
Your errors are related to opencv 4.0, untill we update this issues for the next aerostack release, i suggest you to use an older version of opencv with the command:
sudo apt -y --allow-downgrades install libopencv-dev=3.2.0+dfsg-4ubuntu0.1
You should then roll back to the original code, that you modified in #70 and everything should be working just fine.

ibrahimqazi, 2020-11-24:
> Your errors are related to opencv 4.0, untill we update this issues for the next aerostack release, i suggest you to use an older version of opencv with the command:
> sudo apt -y --allow-downgrades [...]

Thank you for your answer, so which version OpenCV is compatible with your stack ? Thank you

ibrahimqazi, 2021-02-17:
> Your errors are related to opencv 4.0, untill we update this issues for the next aerostack release, i suggest you to use an older version of opencv with the command:
> sudo apt -y --allow-downgrades [...]

@pasanci hello, any updates about this issue, does it work with OpenCV4? or which version is compatible with aerostack? I am stuck here.
Thank you

pasanci, 2021-02-17:
@ibrahimqazi as i mentioned in this response, opencv 3.2.0 should work, you must install it with the command provided.

> Your errors are related to opencv 4.0, untill we update this issues for the next aerostack release, i suggest you to use an older version of opencv with the command:
> sudo apt -y --allow-downgrades [...]

ibrahimqazi, 2021-02-24:
@pasanci thank you but this command does work.
`sudo apt -y --allow-downgrades install libopencv-dev=3.2.0+dfsg-4ubuntu0.1
[sudo] password for ibrrahim: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libopencv-dev is already the newest version (3.2.0+dfsg-4ubuntu0.1).
The following packages were automatically installed and are no longer required:
  gyp libc-ares2 libgflags-dev libgflags2.2 libgoogle-glog-dev
  libgoogle-glog0v5 libhttp-parser2.7.1 libjs-async libjs-inherits
  libjs-node-uuid libmbim-glib4 libmbim-proxy libqmi-glib5 libqmi-proxy
  linux-hwe-5.4-headers-5.4.0-51 linux-hwe-5.4-headers-5.4.0-52
  linux-hwe-5.4-headers-5.4.0-53 linux-hwe-5.4-headers-5.4.0-58
  linux-hwe-5.4-headers-5.4.0-60 node-abbrev node-ansi node-ansi-color-table
  node-archy node-async node-balanced-match node-block-stream
  node-brace-expansion node-combined-stream node-concat-map node-cookie-jar
  node-delayed-stream node-forever-agent node-form-data node-fs.realpath
  node-fstream node-fstream-ignore node-github-url-from-git node-glob
  node-graceful-fs node-hosted-git-info node-inflight node-inherits node-ini
  node-isexe node-json-stringify-safe node-lockfile node-lru-cache node-mime
  node-minimatch node-mkdirp node-mute-stream node-node-uuid node-nopt
  node-npmlog node-once node-osenv node-path-is-absolute node-pseudomap
  node-qs node-read node-request node-retry node-rimraf node-semver node-sha
  node-slide node-spdx-correct node-spdx-expression-parse
  node-spdx-license-ids node-tar node-tunnel-agent node-underscore
  node-validate-npm-package-license node-which node-wrappy node-yallist nodejs
  nodejs-doc usb-modeswitch usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.`

I still get the compilation error.

pasanci, 2021-02-24:
@ibrahimqazi The command has properly installed the required package.
However since the problem continues, all i can think about is trying to completely remove possible opencv remains with the commands :
  sudo apt-get purge '*opencv*'

And:

  sudo find / -name "*opencv*" -exec rm -rf {} \;

Then once again install the required version with:

  sudo apt -y --allow-downgrades install libopencv-dev=3.2.0+dfsg-4ubuntu0.1

If this does not fix the issue, i can only suggest you to try with a fresh ubuntu 18 and ros melodic installation, which is guaranteed to work out of the box.